lgsvl/simulator #1123: 128 ray lidar performance

BaltashovIlia, 2020-11-10:
Hello LGSVL Team, 

I have an issue with performance of simulator using 128 ray lidar. 

I created a vehicle with example json for Velodyne VLS-128 and got lidar data only on 6 Hz and also the simulator freezes.

As I can see in htop one thread has 100% load when using 128 ray lidar. So I think the problem can be either somewhere in lidar data pipeline in simulator because it uses only one thread for some part of the task or GPU acceleration for lidar is not used in mine. 

My PC is AMD Ryzen 7 2700X + Nvidia RTX 2070. 
Without lidar:
[image]
With lidar:
[image]

EricBoiseLGSVL, 2020-11-10:
@BaltashovIlia Yes, 128 lidar uses tons of system memory and also is difficult to send that amount of data over the bridge.  We created a Velodyne lidar plugin that directly communicates with the velodyne driver.  We also are working on raytracing support for our current lidar sensor that will greatly improve performance.

tharindurmt, 2020-11-11:
@EricBoiseLGSVL Is there any plans to release the code for the Velodyne lidar plugin?

hadiTab, 2020-11-11:
@tharindurmt I don't know about plans for releasing the sourced code for the Velodyne lidar, but do you need the source code? The AssetBundle is prebuilt and included in the release binary. You can add the sensor to your vehicle configuration and use it (see [documentation](https://www.lgsvlsimulator.com/docs/lidar-plugin/#velodyne-lidar-sensor-plugins)).

tharindurmt, 2020-11-12:
@hadiTab Thanks for the info. Yes, I do need the source code for the Velodyne lidar plugin as I want it to be modified and be used with another simulator (unfortunately, a requirement of the project that I'm working on).

lemketron, 2020-11-13:
@tharindurmt 
> Yes, I do need the source code for the Velodyne lidar plugin as I want it to be modified and be used with another simulator (unfortunately, a requirement of the project that I'm working on).

The source code for the Velodyne plug-in is not part of our open source release, and just to clarify, the software license agreement does not allow simulator source code to be used in a commercial product without first obtaining a separate license -- but if that is something you're interested in, please contact us at "contact@lgsvlsimulator.com" to discuss further.

tharindurmt, 2020-11-13:
@lemketron Thanks for the detailed info and the clarification.
The project I'm working on is a research project and the constraints that we have are due to some other requirements (it's a multi-university collaboration). 
I'll definitely contact you if our requirements (and/or research directions) get changed.

wenjing1170, 2022-10-06:
@BaltashovIlia Hi, I'd like to ask where you find the  Velodyne VLS Lidar sensor plugin.
I want to use the velodyne lidar sensor plugin and follow the [doc](https://www.svlsimulator.com/docs/archive/2020.06/lidar-plugin/). But I can not find the built asset bundle of this plugin or asset in the source version of LGSVL simulator. I have tried multiple LGSVL simulator versions, and can not find the Velodyne Lidar Sensor plugin. 
Thanks you very much!

wenjing1170, 2022-10-07:
@EricBoiseLGSVL Hi, I want to use velodyne lidar sensor. I config the VLP_32 lidar sensor json following official doc in both built binary LGSVL and source built LGSVL. But I got the error: Exception: Unknown VelodyneLidarType parameter for Velodyne VLP-32C sensor on Jaguar2015XE vehicle. 
Thanks.

EricBoiseLGSVL, 2022-10-07:
velodyne sensor requires 2021.3 release

wenjing1170, 2022-10-08:
@EricBoiseLGSVL I really appreciate your reply.  I am indeed using the 2021.3 version simulator.  For the built binary 2021.3 release, there is no built asset bundle  of the Velodyne Lidar Plugin.  

I modify the JSON file based on the default Lidar sensor configuration JSON file following the [official doc](https://www.svlsimulator.com/docs/plugins/lidar-plugin/#velodyne-lidar-sensor-plugins). The VLP_32C sensor configuration are as follows:
  {
    "params": {
      "LaserCount": 32,
      "MinDistance": 0.5,
      "MaxDistance": 100,
      "RotationFrequency": 10,
      "MeasurementsPerRotation": 1800,
      "FieldOfView": 41.33,
      "CenterAngle": 10,
      "Compensated": true,
      "PointColor": "#ff000000",
      "Topic": "/points_raw",
      "Frame": "velodyne",
      "VelodyneLidarType": "VLP_32C",
      "HostName": "10.37.1.236",
      "UdpPortData": 2368,
      "UdpPortPosition": 8308
    },
    "transform": {
      "x": 0,
      "y": 2.312,
      "z": -0.3679201,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    },
    "name": "Velodyne VLP-32C",
    "parent": null,
    "pluginId": "b30d0478-8c7b-4687-bfc2-b3cdb3f5faff",
    "sortKey": 3,
    "plugin": {
      "isFavored": true,
      "isShared": false,
      "isOwned": false,
      "accessInfo": {
        "userAccessType": "favored",
        "owner": {
          "id": "0d888b00-fa53-47c1-882a-b68391268a11",
          "firstName": "SVL",
          "lastName": "Content"
        }
      },
      "supportedSimulatorVersions": [
        "2021.3",
        "2021.2",
        "2021.2.2",
        "2021.1",
        "2021.1.1",
        "Development"
      ],
      "id": "b30d0478-8c7b-4687-bfc2-b3cdb3f5faff",
      "name": "Lidar Sensor",
      "type": "LidarSensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "This sensor returns a point cloud after 1 revolution.\nSee https://www.svlsimulator.com/docs/simulation-content/sensors-list/#lidar for more details.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "imageUrl": "/api/v1/assets/download/preview/dd44a969-c038-4966-a39f-a445ab3b6c00",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      },
      "shareRequests": []
    },
    "type": "VelodyneLidar"
  },

 When I run the simulator, the error as follows:
[image]

Thanks for your help!

EricBoiseLGSVL, 2022-10-10:
did you add the sensor to your library?

wenjing1170, 2022-10-11:
@EricBoiseLGSVL Thanks for your reply. I did not add the Velodyne Sensor Plugin to my library. I did not find any related asset or asset bundle of the Velodyne Sensor Plugin.  I just modified the sensor configuration json file based on the available Lidar Sensor Plugin in the library. 
I really appreciate it if you can tell me how to find and add the Velodyne Sensor Plugin (VLP_128).