lgsvl/simulator #1124: How to use LGSVL with Autoware and Python API at the same time.

dohkim119, 2020-11-11:
I'm trying to use lgsvl with Autoware and Python API at the same time. 
However, it seems not work properly.
Is there anyway I can solve this?
(Currently, I'm using LGSVL 2019.09 )

hadiTab, 2020-11-11:
Can you provide more details as to how things are set up? What does your Python API script look like? Did you add a bridge to the ego vehicle?

marcusvinicius178, 2021-04-07:
Hi @dohkim119 I am trying to do the same integration with PythonApi+ AutowareAI or Visual Scene Editor+AutowareAi...however my attempts failed. Did you figure out how to set these frameworks to work together. Is it Possible? If yes could you say me how please? Thanks

lemketron, 2021-04-07:
I would suggest upgrading to the 2021 Simulator and the latest Python API and documentation. 2019 is really old at this point, and our focus from here on out is going to be on 2021, not fixing issues with 2020 or older versions...  (Note that VSE is a 2021-only feature.)