lgsvl/simulator #1125: gps coordinates

naamaStudent, 2020-11-12:
hii,
in python API documentation i see this :
data = gps_sensor.data()
print("Latitude:", data.latitude)
but when i write it in my code it says-
   data = gps_sensor.data()
NameError: name 'gps_sensor' is not defined
 can somone please explain to me why?

luqiang21, 2020-11-12:
@naamaStudent Can you provide the url for the documentation you mentioned?

naamaStudent, 2020-11-12:
https://www.lgsvlsimulator.com/docs/python-api/
Under: SENSORS < GPS Sensors

EricBoiseLGSVL, 2020-11-12:
Did you get a reference to the gps sensor before you try to use it?
```
# get_sensors returns a list of sensors on the EGO vehicle
sensors = ego.get_sensors()
for s in sensors:
    if s.name == "Main Camera":
        # Camera and LIDAR sensors can save data to the specified file path
        s.save("main-camera.png", compression=0)
        s.save("main-camera.jpg", quality=75)
        break
```

naamaStudent, 2020-11-19:
yes i did it but still gives me error..
although i dont understand why it is necessary.

EricBoiseLGSVL, 2020-11-19:
You need a ref so API can access the sensor.  If this is null then you can't get data.  Please check if this is null