lgsvl/simulator #1127: unity editor compile error

Michealhao1991, 2020-11-16:
when  i open unity editor (2020.1.0b4) it has some compile error about "documentation.version"

[image]

martins-mozeiko, 2020-11-16:
That's a wrong Unity version. Required version is in `ProjectSettings/ProjectVersion.txt` file, for example - for 2020.06 branch [here](https://github.com/lgsvl/simulator/blob/release-2020.06/ProjectSettings/ProjectVersion.txt).

If you want to use different version you'll need to upgrade HDRP package manually as Simulator has done some modifications there - check the git [history on folder](https://github.com/lgsvl/simulator/commits/release-2020.06/Packages/com.unity.render-pipelines.high-definition) if you want to see what was modified.

Michealhao1991, 2020-11-16:
@martins-mozeiko  thank you very much

closed 2020-11-19 by EricBoiseLGSVL