lgsvl/simulator #1128: Ego vehicle stucks at intersections

Yimags, 2020-11-16:
Hello guys,

I built Apollo 5.0 make run the modular test with latest lgsvl in San Francisco map. When I test route editing function, I noticed that the ego vehicle will stuck at the transection and will stop from moving no matter what color the traffic light is. I check the cyber_monitor, it seems to me that perception of traffic light works well. So I was wondering where the problem locates and how could I solve it. Thanks in advance!
[image]

lemketron, 2020-11-16:
I think you changed the screen shot -- in the original one you had perception enabled but traffic light disabled. You noted that you were using modular testing, so I just wanted to see if you're using modular testing ONLY for traffic light or for perception too (in which case you should disable perception in Dreamview). 

It looks like the traffic signal message is coming through in cyber_monitor, so if Apollo is not recognizing it, then it could be an indication that the Apollo HD map is lacking the matching signal ID from the simulator environment. Could you go into the Apollo 5.0 (LGSVL fork, I assume) source tree and do a "git fetch" and "git status" to see if you have the latest Apollo HD maps? If "git status" shows that your source needs updating then do a "git pull" to update and try again. I apologize for this confusion but we had to update all of the Apollo HD maps to contain signal IDs that match the signal IDs in the Simulator environments.

Yimags, 2020-11-17:
> I think you changed the screen shot -- in the original one you had perception enabled but traffic light disabled. You noted that you were using modular testing, so I just wanted to see if you're [...]

Thanks for your advice. And it shows like this. Would it be a problem?
[image]

Actually I turned on both traffic and perception module at both simulation. However, the button was automatically turned off from time to time and I didn't find the reason. But it seems to me that in 'module delay' page both trafficligt and perception works well so I just ignore the problem.

I run the test just for verification of the planning function. And it works not well because of the given problem. What else I could do to fix this?

ntutangyun, 2020-11-22:
Hi, @Yimags 

* Please try out the latest Apollo master branch instead of Apollo 5.0 since it's no longer maintained. 
* For modular testing, you don't need to turn on the `Traffic` and `Perception` module of Apollo, since the Simulator will directly send out perception results and traffic light results to cyber RT. You need only turn on `Localization, Transform, Routing, Planning, Control, Prediction`.

Manoj-EK, 2020-11-23:
@Yimags you can have a look at issue #12998

Yimags, 2020-11-24:
> Hi, @Yimags
>
> * Please try out the latest Apollo master branch instead of Apollo 5.0 since it's no longer maintained.
> * For modular testing, you don't need to turn on the `Traffic` and `Perception` [...]

Thx for that information! And a my problem occurs when I run the simulation in San Francisco. No problem takes place when I turn to Borregas Ave and cube town. It seems quite strange to me.

closed 2020-12-10 by Yimags