lgsvl/simulator #1132: question about Lidar Sensor Plugin

ideasplus, 2020-11-19:
Hi, LGSVL Team, 

I want to use lidar Sensor Plugin in Apollo because the detection range of default lidar sensor is limited and morever the FrameRatio of lidar and camera sensors drop to 4hz when I increase the `LaserCount` Parameter from 32 to 128.  

According to previous issue #1123 and [offficiall doc](https://www.lgsvlsimulator.com/docs/lidar-plugin/), I use the Velodyne VLS-128 Sensor Plugin, But I  had some problems and hope to get some help.

My computer is Intel® Xeon(R) Gold 5218 + Quadro RTX 5000 16GB.

First, The performance has not improved sigificantly, the FrameRatio of lidar and camera is still 4hz.
[image]

Second, the perception of NPC is wrong and not match the actual position.
[image]

EricBoiseLGSVL, 2020-11-19:
Are you running Apollo and simulator on the same machine?

ideasplus, 2020-11-20:
> Are you running Apollo and simulator on the same machine?

Yes, I run Apollo and simulator on the same machine.

EricBoiseLGSVL, 2020-11-20:
@luqiang21 Any ideas why? The GPU is more than enough Turning

martins-mozeiko, 2020-11-20:
My guess would be on slow CPU. It is only 2.3GHz. Unity really likes higher clock rate CPUs.

ideasplus, 2020-11-20:
> My guess would be on slow CPU. It is only 2.3GHz. Unity really likes higher clock rate CPUs.

How  can I determine if it is the cpu problem?  What about the second question?

lemketron, 2020-11-20:
@ideasplus You can run `htop` to check the CPU performance and see how busy each of the cores are, and you can use nvidia-smi to check the GPU load.

Please note that just running Apollo in a vehicle can easily demand a pretty high performance machine. It's not just about GPU memory but also clock speed as @martins-mozeiko mentioned. I believe your CPU can run up to 3.9GHz in turbo mode but it's worth checking to see how fast it's actually running. I've found `turbostat` can help by dynamically reporting clock speeds. If it's somehow stuck at 2.3GHz then that could definitely be an issue.

The bottom line is that while there are tricks (like "[Modular Testing](https://www.lgsvlsimulator.com/docs/modular-testing/)") that can reduce some of the demands Apollo places on the CPU and GPU, you will have the best results if you can run Apollo on a separate machine -- especially if you eventually need to add more simulated sensors to your vehicle.

Regarding obstacles showing up in the wrong spot in Dreamview, that sounds like the Apollo map may not be in sync with the environment. Are you using LGSVL Simulator 2020.06? What version of Apollo (and from where)? You should make sure you have the [latest Apollo maps](https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/map/data) so they match the latest 3D environments.

EricBoiseLGSVL, 2020-11-20:
Yeah the clouds look older.  Is this the latest release 2020.06p?  This is a branch, not master

ideasplus, 2020-11-21:
@lemketron @EricBoiseLGSVL Thank you for your suggestions. 

> Regarding obstacles showing up in the wrong spot in Dreamview, that sounds like the Apollo map may not be in sync with the environment. Are you using LGSVL Simulator 2020.06? What version of Apollo [...]

I'm running **LGSVL Simulator 2020.06** with **Apollo pre6**. I don't think the sceond problem is caused by the map because when I use the defalut [Sample sensor configuration](https://www.lgsvlsimulator.com/docs/apollo5-0-json-example/) and it works well.
[image]

> The bottom line is that while there are tricks (like "Modular Testing") that can reduce some of the demands Apollo places on the CPU and GPU, you will have the best results if you can run Apollo on [...]

The `Modular Testing` is very useful **BUT I actuallly want to test the performance of perception moudle.**

> Please note that just running Apollo in a vehicle can easily demand a pretty high performance machine. It's not just about GPU memory but also clock speed as @martins-mozeiko mentioned. I believe [...]

I've used this `indicator-cpufreq ` tool to adjust the cpus to high performance mode. Please correct me if I make mistakes.

Here are some screenshots to better illustrate the problems.

- After running `HTOP`
[image]

- After running `turbostat`
[image]