lgsvl/simulator #1133: run lgsvl over ssh failed

InitialDuan, 2020-11-19:
Hello, I'm trying to run lgsvl on a remote linux machine over ssh. I downloaded binary release and installed vulkan on the remote machine. When I run `./simulator`, it shows the following output:
```
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  91
  Current serial number in output stream:  92
```
Here is the nvidia-smi output of the remote machine:
```
Thu Nov 19 13:25:52 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 435.21       Driver Version: 435.21       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 1070    Off  | 00000000:01:00.0  On |                  N/A |
| 63%   78C    P2   120W / 190W |   5617MiB /  8112MiB |    100%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1825      G   /usr/lib/xorg/Xorg                            33MiB |
|    0      1902      G   /usr/bin/gnome-shell                          67MiB |
|    0     17488      C   python                                      2347MiB |
|    0     19633      C   python                                      3157MiB |
+-----------------------------------------------------------------------------+
```
I didn't have a `unity3d` repository under `./config` on the remote machine, so I can't find the Player.log. Would it automatically create a `unity3d` repository when I run the lgsvl or it's a default repository on Ubuntu?

ntutangyun, 2020-11-19:
does this remote machine have a monitor connected to it ?

InitialDuan, 2020-11-19:
> does this remote machine have a monitor connected to it ?

I am not sure. Actually I want to display it on my local machine, since my local GPU dosen‘t have enough memory, is that possible?  Is the GPU used by lgsvl for “computation“ or “displaying“?

EricBoiseLGSVL, 2020-11-19:
GPU is used for both displaying and computation. We have a few issues here over running sim over SSH but it is not recommended. The GPU on the remote machine is boarder line but will work.  Since we are not familiar with your server setup, it will be hard to debug.  @ntutangyun might be right, a missing monitor could be the cause but it could be many things.  Maybe a user can point you in the right direction.

martins-mozeiko, 2020-11-20:
Running simulator on different machine is not a problem, but you cannot use X11 forwarding over SSH. Because of two reasons 1) it defeats purpose of using GPU on other machine - as it would forward all rendering commands to your GPU; 2) it does not work for modern OpenGL or Vulkan rendering, it works only for old GL.

What you need to do is run real x11 server on remote machine. And then run VNC server to connect from your first machine to see the desktop. No need for monitor. As long as VNC server will capture whatever is displayed on x11, everything will be fine. For performance reasons you might consider using different remote desktop solution, like NoMachine or Chrome Remote Desktop - both of these can send video instead of still pictures, so it will have higher performance / less bandwidth usage.

InitialDuan, 2020-11-22:
@martins-mozeiko Thanks you so much! I just used the remote desktop application like Teamiviewer and it worked well!

closed 2020-11-22 by InitialDuan