lgsvl/simulator #1135: Cloud URL is down, disabling the sim

mjshiggins, 2020-11-20:
It looks like [https://account.lgsvlsimulator.com](https://account.lgsvlsimulator.com) is down, which results in the sim crashing on startup. There should be a fallback to disable the cloud functionality.

Player.log

> SocketException: The requested address is not valid in this context
>   at System.Net.Sockets.Socket.Connect (System.Net.EndPoint remoteEP) [0x00055] in <ae22a4e8f83c41d69684ae7f557133d9>:0
>   at System.Net.WebConnection.Connect (System.Net.HttpWebRequest request) [0x0016d] in <ae22a4e8f83c41d69684ae7f557133d9>:0
> Rethrow as WebException: Error: ConnectFailure (The requested address is not valid in this context)
>   at System.Net.WebClient.DownloadDataInternal (System.Uri address, System.Net.WebRequest& request) [0x00072] in <ae22a4e8f83c41d69684ae7f557133d9>:0
>   at System.Net.WebClient.DownloadData (System.Uri address) [0x00020] in <ae22a4e8f83c41d69684ae7f557133d9>:0
>   at System.Net.WebClient.DownloadData (System.String address) [0x00016] in <ae22a4e8f83c41d69684ae7f557133d9>:0
>   at (wrapper remoting-invoke-with-check) System.Net.WebClient.DownloadData(string)
>   at GoogleAnalyticsMPV3+<>c.<.cctor>b__19_0 () [0x00011] in <a572951013734ceaa8539e215f80cdfa>:0
>   at System.Threading.ThreadHelper.ThreadStart_Context (System.Object state) [0x00014] in <437ba245d8404784b9fbab9b439ac908>:0
>   at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x000$
>   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <$
>   at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <437ba245d8404784b9fbab9b439ac908$
>   at System.Threading.ThreadHelper.ThreadStart () [0x00008] in <437ba245d8404784b9fbab9b439ac908>:0
> UnityEngine.DebugLogHandler:Internal_LogException(Exception, Object)
> UnityEngine.DebugLogHandler:LogException(Exception, Object)
> UnityEngine.Logger:LogException(Exception, Object)
> UnityEngine.Debug:LogException(Exception)
> UnityEngine.<>c:<RegisterUECatcher>b__0_0(Object, UnhandledExceptionEventArgs)

mjshiggins, 2020-11-20:
Update: I have a working version of the simulator that removes the cloud dependency entirely. Will post a link to the branch once it's up.

hadiTab, 2020-11-21:
@mjshiggins Forgot to post this here earlier, but thanks for pointing it out. The URL is backup now.

closed 2020-11-24 by EricBoiseLGSVL