lgsvl/simulator #1136: Lgsvl Map details

asifsid-32, 2020-11-21:
Hello team, 

We have calibrated the fisheye camera sensors by placing calibration patterns in Cubetown Map because it was open source. 

I have few points to confirm -

1) Now when i am driving the ego car in Autonomous Map the results are not the same. Is the Autonomous Stuff Map different than CubeTown with respect to plane ? Are there any random steeps and slopes which are not visible but exist in Autonomous Stuff ? 

2) Are there any closed parking slot available in any of the Maps present ?

ntutangyun, 2020-11-22:
I remember GoMentum has some parking lots also.

EricBoiseLGSVL, 2020-11-24:
@asifsid-32 
1) AutonomouStuff map is not flat.  This map was generated from real world data.
2) We have parking spots in AutonomouStuff and Gomentum.  We are currently annotating parking spots in San Francisco and added 3 different parking lot styles.  This will release soon.

closed 2020-12-02 by EricBoiseLGSVL