lgsvl/simulator #1139: more them two crosswalk cannot be shown on dreamview?

Addi2020, 2020-11-23:
I have created 9 crosswalk (zebra crossing) in one intersection and junction, as you can see the picture 1. 
But when upload HD map to dream view it only shows two crosswalk? (picture 2)

picture 1. 
[image]

picture 2. 

[image]

luqiang21, 2020-11-23:
@Addi2020 I suggest you to first check how many crosswalks were exported. To visually check, you can convert the `.bin` file back to `.txt`. After that, you can know where the issue is. Refer to [here](https://github.com/lgsvl/simulator/issues/991#issuecomment-695219933) for related commands.

Addi2020, 2020-11-24:
i convert .bin to .txt and find out that there are 9 crosswalk but only crosswalk(0) and crosswalk(9) contains id "CW_0", "CW_9", other do not having id. 

crosswalk {
  id {
    id: "CW_0"
  }
  polygon {
    point {
      x: 802156.31019592285
      y: 2489742.3306846619
    }
    point {
      x: 802156.803024292
      y: 2489747.8772163391
    }
    point {
      x: 802161.5979309082
      y: 2489747.7262496948
    }
    point {
      x: 802161.17724609375
      y: 2489742.1796913147
    }
  }
  overlap_id {
    id: "overlap_CW_0_lane_55"
  }
}
crosswalk {
  polygon {
    point {
      x: 802181.84893798828
      y: 2489778.12758255
    }
    point {
      x: 802186.5439453125
      y: 2489777.8536720276
    }
    point {
      x: 802186.40200805664
      y: 2489772.8962898254
    }
    point {
      x: 802181.17861938477
      y: 2489772.693447113
    }
  }
}
crosswalk {
  polygon {
    point {
      x: 802187.38525390625
      y: 2489765.231716156
    }
    point {
      x: 802191.35968017578
      y: 2489769.139957428
    }
    point {
      x: 802196.58639526367
      y: 2489764.1503105164
    }
    point {
      x: 802192.70928955078
      y: 2489759.8801994324
    }
  }
}
crosswalk {
  polygon {
    point {
      x: 802196.64834594727
      y: 2489756.3928794861
    }
    point {
      x: 802200.6227722168
      y: 2489760.3011207581
    }
    point {
      x: 802205.84948730469
      y: 2489755.3114738464
    }
    point {
      x: 802201.9723815918
      y: 2489751.0413627625
    }
  }
}
crosswalk {
  polygon {
    point {
      x: 802166.69575500488
      y: 2489748.9696464539
    }
    point {
      x: 802162.82720947266
      y: 2489753.6364212036
    }
    point {
      x: 802167.82661437988
      y: 2489759.2310714722
    }
    point {
      x: 802171.71865844727
      y: 2489754.4311790466
    }
  }
}
crosswalk {
  polygon {
    point {
      x: 802174.382232666
      y: 2489759.1781578064
    }
    point {
      x: 802170.74681091309
      y: 2489763.96018219
    }
    point {
      x: 802175.74612426758
      y: 2489769.5548362732
    }
    point {
      x: 802179.40380859375
      y: 2489764.3486404419
    }
  }
}
crosswalk {
  polygon {
    point {
      x: 802170.845199585
      y: 2489738.326259613
    }
    point {
      x: 802166.54891967773
      y: 2489734.1510505676
    }
    point {
      x: 802161.703048706
      y: 2489739.4346580505
    }
    point {
      x: 802166.01585388184
      y: 2489743.252117157
    }
  }
}
crosswalk {
  polygon {
    point {
      x: 802179.3642578125
      y: 2489729.7279968262
    }
    point {
      x: 802175.06799316406
      y: 2489725.5526275635
    }
    point {
      x: 802168.13766479492
      y: 2489733.0000419617
    }
    point {
      x: 802172.45050048828
      y: 2489736.8175544739
    }
  }
}
crosswalk {
  polygon {
    point {
      x: 801947.648563385
      y: 2489536.0054321289
    }
    point {
      x: 801948.66073036194
      y: 2489541.0308074951
    }
    point {
      x: 801954.2982635498
      y: 2489540.7686920166
    }
    point {
      x: 801953.02240753174
      y: 2489535.1062927246
    }
  }
}
crosswalk {
  id {
    id: "CW_9"
  }
  polygon {
    point {
      x: 801973.95249938965
      y: 2489564.2182769775
    }
    point {
      x: 801979.26593780518
      y: 2489563.5351867676
    }
    point {
      x: 801978.89517211914
      y: 2489557.9031677246
    }
    point {
      x: 801972.94956970215
      y: 2489558.5475006104
    }
  }
  overlap_id {
    id: "overlap_CW_9_lane_56"
  }
}

luqiang21, 2020-11-24:
@cyberphysicalrobotics Do you know why some of the crosswalks have ids while others don't?

Addi2020, 2020-11-26:
I find out the issue and resolved.
First we need to do map annotation. crosswalk need annotated lanes, that overlap the crosswalk. after adding lanes , that issue is solved.

closed 2020-11-26 by Addi2020