lgsvl/simulator #1140: About cyber channel /apollo/sensor/gnss/heading in Apollo 5.0

AlexandrZabolotny, 2020-11-23:
Hello team!

It looks like a channel "**/apollo/sensor/gnss/heading**" is needed for the msf_localization to work correctly.
This channel forms the GNSS driver (for novatel).

Can your simulator create this channel?
Thanks!

ntutangyun, 2020-11-24:
if you're using the simulator, may I know why you want to use `msf_localization` instead of `rtk_localization`?

also, you could create a custom sensor [(check here)](https://www.lgsvlsimulator.com/docs/sensor-plugins/) to publish this channel if necessary.

AlexandrZabolotny, 2020-11-24:
> if you're using the simulator, may I know why you want to use `msf_localization` instead of `rtk_localization`?
> 
> also, you could create a custom sensor [(check here)](https://www.lgsvlsimulator.com/docs/sensor-plugins/) to publish this channel if necessary.

I want to understand how MSF localization works in Apollo using a simulator.

I can't run MSF localization on a real car, so I'm trying to figure it out on a simulator

closed 2020-11-24 by AlexandrZabolotny

mengen100, 2021-04-09:
@AlexandrZabolotny Hi Alex, I also want to use MSF localization module for my simulation, have you figured how to make Apollo control the vehicle in LGSVL with MSF localization module? 
@ntutangyun Thanks for the reply earlier, I'd like to understand more about msf_localization if possible. From what I've tried, routing would not work without enabling the localization module in dreamview, but if we enable it, how do we tell the vehicle is locating based on **localization module we enabled in dreamview** (which should be ndt in this case) or the **MSF localization module** we started in the command line?

AlexandrZabolotny, 2021-04-11:
> @AlexandrZabolotny Hi Alex, I also want to use MSF localization module for my simulation, have you figured how to make Apollo control the vehicle in LGSVL with MSF localization module?
> @ntutangyun [...]

Not.
I am trying to do this on a real car, where the gnss driver generates a full set of required channels

Juntu-hub, 2021-06-22:
Hi, Alexandr. @AlexandrZabolotny Are you sure gnss/heading is necessary for msf_localization? I am also doing experiments on apollo with lgsvl. I found the msf_lidar is off and on when I run simulator with the local_map built by recorders recorded in lgsvl also. Do you know what the problem is? Thank you.