lgsvl/simulator #1143: Traffic light direction is wrong in Apollo map

qiongwu1003, 2020-11-24:
Hi,

simulator : 202006
Apollo : 5.0

I made a simple 3D map and HD map reference [here](https://www.lgsvlsimulator.com/docs/map-annotation/#create-traffic-signals), it shows well in the simulator. But when I generate the HD map to Apollo according to this [issue](https://github.com/lgsvl/simulator/issues/370), some traffic lights directions are wrong. Am I did something wrong?

Below is the one error traffic light and its parameter.
[image]
[image]
[image]
[image]

luqiang21, 2020-11-24:
@qiongwu1003 Can you show us the annotations of the signal?

qiongwu1003, 2020-11-24:
Do you mean this one:
[image]
[image]

luqiang21, 2020-11-24:
@qiongwu1003 Yes, are your signals having the same directions as these annotations?

qiongwu1003, 2020-11-24:
Yes, they are. Z axis forward and Y axis up.
[image]

EricBoiseLGSVL, 2020-11-25:
@qiongwu1003 Can you please screen shot the intersection so we can see if the signal and stopline objects are facing the correct way?  Be sure Toggle Tool Handle Rotation is in Local Space

Like so
[image]
[image]

qiongwu1003, 2020-11-25:
@EricBoiseLGSVL 
Thanks for your reply, below are the screenshot.
[image]
[image]

EricBoiseLGSVL, 2020-11-26:
Hmm, that is right,  Very odd that the signals are inverted.

qiongwu1003, 2020-11-26:
@EricBoiseLGSVL 
I also found that the traffic lights' directions are wrong in some intersections of the Sanfrancisco map.
Maybe the same error with my HD map? Or my computer's problem, setting or something? Thanks a lot.
[image]

EricBoiseLGSVL, 2020-11-26:
Can you screen shot the intersection please.  It will help to debug both issues

qiongwu1003, 2020-11-26:
It is “MapIntersection_junction_89_0” in Sanfrancisco HD map.
The "MapSignal_signal_124" is wrong.
[image]

The "MapSignal_signal_122" is correct, but I cant find out the difference.
[image]

EricBoiseLGSVL, 2020-11-30:
Thanks for posting.  These all look correct. Maybe this is an issue with exporting the HDmap.  I'll ping @luqiang21 and see if he knows how to check the export.