lgsvl/simulator #1144: CanBus data logging

nike-96, 2020-11-24:
Hi, is there the possibility of getting data of the CanBus sensor on Python API? I tried the attribute "data" which is available for the GPS, but it's not an attribute of the object CanBusSensor.

EricBoiseLGSVL, 2020-11-25:
@nike-96 Yes, I have MRs in review now and it will be in our next release.  I can post the patch files here if you don't want to wait

nike-96, 2020-11-26:
Don't worry, I can wait. Thank you very much

EricBoiseLGSVL, 2021-03-03:
Sorry this issue slipped through the cracks.  Sorry for the confusion. After reviewing all the sensors that would need to be added and the performance issues, we decided to not support api for data collection. Please use the bridge for data collection. You can change the source code to add canbus but we do not recommend. You will need to change the python api and api commands in simulator source.  Maybe @cubrink would be able to work with you from this issue #1278

closed 2021-03-03 by EricBoiseLGSVL