lgsvl/simulator #1147: lgsvl_msgs/VehicleControlData confusion

soldierofhell, 2020-11-25:
Hi,
I use this message to control vehicle from ROS2, but I'm confused by the fact that it has acceleration_pct as input. It's seems to me that this is really a throttle pct, right? I observe exacly the same value in the output lsgvl_msgs/CanBusData in the throttle_pct field

By the way, for acceleration steering it would be best to use commonly understood messages in ROS2 community like http://docs.ros.org/en/melodic/api/ackermann_msgs/html/msg/AckermannDrive.html instead of a custom message, what you think?

Regards,

EricBoiseLGSVL, 2020-11-25:
@soldierofhell The reason we have a custom message is from our simple VehicleSMI.cs that controls dynamics.  If you implement a more complex dynamics class then this would be more appropriate.  Right @hadiTab ?

hadiTab, 2020-11-25:
@EricBoiseLGSVL is right. To add to it the simulator does not directly apply a desired acceleration, speed, etc. to the ego vehicle. The  vehicle dynamics model, coupled with the physics engine will determine what acceleration or speed the ego vehicle achieves for a certain throttle/brake and steer value. The user will generally want to have a closed-loop controller in ROS to ensure the desired acceleration, speed, etc. is achieved.

@soldierofhell You are right, the acceleration_pct is really throttle_pct.

soldierofhell, 2020-11-26:
> @soldierofhell You are right, the acceleration_pct is really throttle_pct.

@EricBoiseLGSVL, @hadiTab Thanks guys for confirmation. I'd suggest to at least add this as a comment in the message, because acceleration is typically an output from hight level controller (e.g. mpc) and my initial thinking was 'great, they did low-level control internally, I don't have to deal with the calibration', which is not true.

BTW IMO the target_wheel_angle is wrong. ROS/ROS2 convention is that steering is positive to the right, see e.g.
https://github.com/osrf/car_demo/blob/master/prius_msgs/msg/Control.msg