lgsvl/simulator #1149: Error:“Type is not expected, and no contract can be inferred: People.person”

ilovevision, 2020-11-26:
The "person.cs" of "protobuf-net-2.4.0\src\Examples\ProtoGen\person.proto" by "protoc  --csharp_out=./ person.proto"  is different from "protobuf-net-2.4.0\src\Examples\person.cs by protogen"

My version of protoc is libprotoc 3.14.0, and the person.cs is follow
[person.pdf](https://github.com/lgsvl/simulator/files/5602046/person.pdf)

My test：
var person = new People.person
{
    Id = 1001,
    Name = "Fred",
    Email = "123456@google.com",
};
using (var file = File.Create("person.bin"))
{
    Serializer.Serialize(file, person);
}

People.person newPerson;
using (var file = File.OpenRead("person.bin"))
{
    newPerson = Serializer.Deserialize<People.person>(file);
}

but  it throw a error,as follows:
System.InvalidOperationException:“Type is not expected, and no contract can be inferred: People.person”

I don't to konw why.

EricBoiseLGSVL, 2020-12-01:
Yes lgsvl protobuf is different, I am confused on what your issue is.

daviduhm, 2020-12-02:
@ilovevision Different versions of protoc library might result in different cs outputs. We use our forked/modified version of protobuf-net to generate cs files for Apollo "specific" to our simulator. You can refer to [this](https://github.com/lgsvl/simulator/issues/207) if you're trying to figure out how to generate protobuf files for the simulator. Please use the [protobuf-net](https://github.com/protobuf-net/protobuf-net) repo for any general questions about protobuf messages that are not related to the simulator.