lgsvl/simulator #1150: Read and save Radar and IMU data from bridge

Yimags, 2020-11-26:
Hello all,

I run the simulator together with Apollo and try to get the data collected by different sensors. After going through the documentation I found that the way to save camera and Lidar information is well documented, but not the method to get data from Radar, IMU and CanBus etc. So how could I get these information from cyberRT? 

Thanks in advance.

EricBoiseLGSVL, 2020-12-01:
@Yimags We are adding the ability to get data with the api for these sensors.  Is Radar, IMU and CanBus the only sensors needed?

Yimags, 2020-12-01:
> @Yimags We are adding the ability to get data with the api for these sensors.  Is Radar, IMU and CanBus the only sensors needed?

Yeah, these will be enough for me. Thx!

Yimags, 2020-12-01:
> @Yimags We are adding the ability to get data with the api for these sensors.  Is Radar, IMU and CanBus the only sensors needed?

@EricBoiseLGSVL btw, how could I subscribe topics and save relevant data, in, for example, txt format? Considering the API is not available for now...

EricBoiseLGSVL, 2020-12-01:
traffic light is not a sensor.  It would be better to use the bridge and save the data with that.  This is the way we designed it for data collection.

Yimags, 2020-12-11:
> traffic light is not a sensor. It would be better to use the bridge and save the data with that. This is the way we designed it for data collection. Using the API to get this data will be very slow.
> [Here](https://www.lgsvlsimulator.com/docs/lgsvl-msgs/) is the link to use lgsvl messages.

Is it possible to use bridge (cyberRT) to get these data within Python API? Actually I am considering using Python API together with both lgsvl and apollo to save data when running simulations. Would you provide me some more information about how to integrate Python API and CyberRT-based data collection?

EricBoiseLGSVL, 2020-12-11:
So the bridge is used with the python API.  Api is just to control the simulation while the sensors send the data through the optimized bridge.  Run the bridge, listen to topics and you are good.  Right @hadiTab ?

hadiTab, 2020-12-11:
Pretty much. You can use the Cyber recorder to save data in Apollo.

Yimags, 2020-12-11:
> Pretty much. You can use the Cyber recorder to save data in Apollo.

Thx for the brief guidance! And here is the case I came across with when I try to run with cyber_recorder. Could you point me out about how could I fix it?
[image]

EricBoiseLGSVL, 2020-12-21:
@Yimags after creating data collection with the API we found many issues with performance.  We will not be releasing get data API commands because the bridge was designed to do this.  As for the error, @hadiTab or @luqiang21 do you know why the cyber recorder is having this error?  Is this still an issue @Yimags ?

closed 2020-12-21 by EricBoiseLGSVL

Yimags, 2020-12-21:
> @Yimags after creating data collection with the API we found many issues with performance. We will not be releasing get data API commands because the bridge was designed to do this. As for the [...]

Thx for the information!
And my issue has been solved [here](https://github.com/ApolloAuto/apollo/issues/13178#issue-764275547).

EricBoiseLGSVL, 2020-12-21:
Excellent.  Glad it was solved