lgsvl/simulator #1151: Issues adding new map to lgsvl simulator

MartinaB85, 2020-11-27:
Good morning,

I would need to add a new map to the simulator. I have Unity version 2019.3.15 running on Vulkan and 18.04 Ubuntu version.
The scenario I'm trying to add  was made with the same version of Unity on Windows, using prefabs downloaded from the Unity asset store. When I import it into a new scene in the simulator, some of the textures (roads, vegetation, terrain) do not show, as you can see in the picture in attached. By opening it with OpenGL API it does work but then OpenGL is not supported by the simulator. So how can I fix this problem with Vulkan API?

Many thanks in advance for your help,

Martina

[image]

EricBoiseLGSVL, 2020-11-30:
@MartinaB85 Welcome to LGSVL Simulator.
Unfortunately, many assets on the asset store are not HDRP compatible. Fortunately, there is an easy fix.
- Open the asset in a non HDRP project.
- Organize all assets into the Assets/External/Environments/Brennero (it will make it easier later)
- Go through the shaders used in the asset purchased.  Change all materials to Unity's standard shaders.
- Remove all custom shaders from the purchased asset.  All assets should have textures and use Unity's standard shader
- Right click the folder and Export custom package.
- Open simulator
- Right click and import custom package.
- Select all materials in the newly imported package.
- Edit -> RenderPipeline -> Upgrade Selected materials to HDRP materials.
- Tweak the materials to add transparency, change material properties as needed.
Now all meshes should be visible.

MartinaB85, 2020-12-07:
Thank you very much. The scenario was made with custom shaders, so I guess the only way is to change all of them manually :(

EricBoiseLGSVL, 2020-12-07:
Yes, it is best to convert before importing and then use unity's built in converter to HDRP materials.

closed 2020-12-07 by EricBoiseLGSVL