lgsvl/simulator #1154: Problems with the planning module on apollo

Turi30, 2020-11-30:
Hi,
I'm trying to run apollo with the lgsvl simulator. The apollo host have Ubuntu 18.04 with 32GB RAM, Intel Core i7-6700K, Nvidia GeForce GTX 1080Ti with Driver Version: 450.80 and CUDA Version: 11.0.
I follow all these [instructions](https://github.com/lgsvl/apollo-5.0#launching-apollo-alongside-the-simulator), but sometimes an error occurs. When I start the modules from dreamview, they start. But after I select a destination, the planning module stop.
This happens 3 times over 4 that I try to start apollo. 

Attach [here](https://github.com/lgsvl/simulator/files/5619139/log.zip) the log of the simulation.

lemketron, 2020-11-30:
Which version of Apollo are you using and what are you trying to do in Apollo? If it's general Apollo issues you might ask on the Apollo forum: https://github.com/apolloauto/apollo/issues 

You might also check the documentation site, especially if you are running Apollo master or the LGSVL fork of Apollo 5.0 : https://www.lgsvlsimulator.com/docs/

Turi30, 2020-12-01:
The version of Apollo is 5.0 and I want only run Apollo with the simulator

lemketron, 2020-12-01:
Are you running simulator and apollo on the same machine? If so then it could be that the planning module is crashing for some reason (most probably out of GPU memory). 

Try using the nvidia-smi tool by typing `watch -n1 nvidia-smi` and see if the apollo module starts and runs for a while and then terminates. You can also watch the numbers in the middle "memory usage" box, to get an idea if the GPU is running out of memory.  If it is then you'll need to use less memory. Running dreamview in Chrome on the same system can use GPU memory, and loading some large environments in the simulator also use GPU memory. 

Other than that the two options are to run Apollo on a separate machine from the simulator, or you can avoid running the perception and traffic light modules in Apollo by adding ground truth sensors in the simulator as described in the [Modular Testing documentation](https://www.lgsvlsimulator.com/docs/modular-testing/).

lemketron, 2020-12-01:
> Attach [here](https://github.com/lgsvl/simulator/files/5619139/log.zip) the log of the simulation.

This is from your planning.INFO log:

```
E1130 23:18:43.533604  2754 reference_line_provider.cc:193]  Routing is not ready.
E1130 23:18:43.583727  2757 reference_line_provider.cc:193]  Routing is not ready.
E1130 23:18:43.633878  2757 reference_line_provider.cc:193]  Routing is not ready.
E1130 23:18:43.637238  2752 planning_component.cc:187]  routing not ready; skip the planning cycle.
E1130 23:18:43.637393  2752 planning_component.cc:129]  Input check failed
I1130 23:18:43.646450  2758 planning_component.cc:57]  Received routing data: run routing callback.timestamp_sec: 1606774723.6450298
module_name: "routing"
sequence_num: 0
E1130 23:18:43.683986  2758 reference_line_provider.cc:193]  Routing is not ready.
E1130 23:18:43.734112  2752 reference_line_provider.cc:193]  Routing is not ready.

```

And this is from mine (Apollo 6 using Modular Testing). The log doesn't seem terribly useful (doesn't really indicate that it's working correctly or crashing but you can see how our logs differ, as mine mentions "reference line" and later has messages about switching stage):

```
E1201 12:08:32.615031  2549 reference_line_provider.cc:196] Routing is not ready.
E1201 12:08:32.673063  2554 reference_line_provider.cc:196] Routing is not ready.
E1201 12:08:32.689182  2550 planning_component.cc:246] routing not ready; skip the planning cycle.
E1201 12:08:32.689234  2550 planning_component.cc:157] Input check failed
I1201 12:08:32.691025  2553 planning_component.cc:67] Received routing data: run routing callback.timestamp_sec: 1606853308.2243488
module_name: "routing"
sequence_num: 0
E1201 12:08:32.733614  2563 reference_line_provider.cc:196] Routing is not ready.
W1201 12:08:32.756726  2561 reference_line_provider.cc:257] Reference line is NOT ready.
E1201 12:08:32.756742  2561 reference_line_provider.cc:259] Failed to use reference line latest history
W1201 12:08:33.167925  2564 channel_buffer.h:70] channel[/apollo/prediction] read buffer overflow, drop_message[1] pre_index[71] current_index[72]
W1201 12:08:33.263976  2564 channel_buffer.h:70] channel[/apollo/prediction] read buffer overflow, drop_message[1] pre_index[73] current_index[74]
...
I1201 12:09:09.030973  2552 scenario.cc:90] switch stage from STOP_SIGN_UNPROTECTED_PRE_STOP to STOP_SIGN_UNPROTECTED_STOP
W1201 12:09:09.034116  2552 channel_buffer.h:70] channel[/apollo/prediction] read buffer overflow, drop_message[1] pre_index[430] current_index[431]
W1201 12:09:09.401463  2552 channel_buffer.h:70] channel[/apollo/prediction] read buffer overflow, drop_message[2] pre_index[433] current_index[435]
W1201 12:09:09.575120  2552 channel_buffer.h:70] channel[/apollo/prediction] read buffer overflow, drop_message[1] pre_index[436] current_index[437]
W1201 12:09:09.786970  2552 channel_buffer.h:70] channel[/apollo/prediction] read buffer overflow, drop_message[1] pre_index[438] current_index[439]
I1201 12:09:10.149873  2552 scenario.cc:90] switch stage from STOP_SIGN_UNPROTECTED_STOP to STOP_SIGN_UNPROTECTED_CREEP
...
```