lgsvl/simulator #1155: How to get ladar sensor point cloud data

qiongwu1003, 2020-12-01:
Hi,
LGSVL team

I want to use Velodyne Lidar and get its point cloud data. I see that I can use Python API get lidar data.
And except Python API is there other method get the lidar data, eg some log files.
Or can i add some code to get the lidar data? Thanks.

EricBoiseLGSVL, 2020-12-01:
@qiongwu1003 Velodyne Lidar code is not open source you will not be able to edit.  The velodyne lidar is also able to publish directly to the driver.  You can set the address and port with the sensor parameters
```
```
        [SensorParameter]
        public string HostName = ""; // IP Address for Velodyne Driver

        [SensorParameter]
        public int UdpPortData = 2368; // UDP port for Velodyne Driver data packet

        [SensorParameter]
        public int UdpPortPosition = 8308; // UDP port for Velodyne Driver position packet
```
```

lemketron, 2020-12-01:
The Velodyne LiDAR sensor publishes data over the network just like a real Velodyne sensor. You should be able to use any available driver or software to read this data.

You could also check out the [VeloView LiDAR viewer](https://www.paraview.org/VeloView) software. 

The source code for VeloView is on github here: https://github.com/Kitware/VeloView

qiongwu1003, 2020-12-02:
@lemketron @EricBoiseLGSVL Thanks for your reply. 

1. Where do I add the sensor parameters, LidarSensorBase.cs?

2. I see [here](https://www.lgsvlsimulator.com/docs/lidar-plugin/), there is a velodyne driver in Apollo 5.0.  So If I can running simulator adding velodyne lidar with Apollo 5.0, and using VeloView LiDAR viewer(lemketron recommend) to listen and record the point cloud data?

3. I see another method "If you need continuous stream of large data like camera image or lidar point cloud it is better to use ROS or Cyber messages to record them to rosbag" in this [issue](https://github.com/lgsvl/simulator/issues/168). Could you give some detail tutorial about this method？

Thanks a lot, looking forward to your reply.

qiongwu1003, 2021-01-18:
No further reply, closed.

closed 2021-01-18 by qiongwu1003