lgsvl/simulator #1157: Multiple NPC/Pedestrian triggers not working well

ffzero1, 2020-12-02:
Hello,

I'm 202006 LGSVL simulator version, with latest Python API. In my script, I have multiple NPCs and Pedestrians triggered by EGO - using below lines in each function. The issue is, when I run one NPC or Ped, trigger works perfectly fine. However, when I have multiple NPCs, Peds, to trigger off of  distance from EGO, only 2 works. Is there anything I'm doing wrong here?

Pedestrian:
wp = lgsvl.WalkWaypoint(position, idle=0, trigger_distance=tr, speed=speed, trigger=None)
<uniquePED>.follow(waypoints)

NPC:
wp = lgsvl.DriveWaypoint(lgsvl.Vector(x_pos, y_pos, z_pos), speed, NPCRTAngle, idle=0, trigger_distance=tr, trigger=None)
<uniqueNPC>.follow(waypoints)

EricBoiseLGSVL, 2020-12-02:
It shouldn't matter how many agents you are using.  Any idea @hadiTab ?  Doesn't the trigger parameter need a value?

ffzero1, 2020-12-02:
Hello Eric, Thanks for the reply. I do assign values to those variables (position, x_pos, y_pos, z_pos, speed, and tr)..

vatanaksoytezer, 2020-12-17:
@EricBoiseLGSVL I have a similar problem as well with NPCs even without the triggers. I found that while following the waypoints via multiple NPCs other NPCs usually (there are some cases they don't but I have no idea) stop their movement after the first (or first few) vehicles finish their waypoint following. Like @ffzero1 I also made sure that the NPCs work fine on their own, without other NPCs. I can provide the list of waypoints and my script to move the NPCs if you would like.

EricBoiseLGSVL, 2020-12-17:
It maybe an issue with distance to ego.  Any data to help reproduce the issue would be great.

vatanaksoytezer, 2020-12-17:
The code is a lot messy, as I was kinda in hurry for a demo, please pardon that :). Here in the zip file the main file video_full_test.py which reads some data from the txt given and does some conversions to calculate waypoints, and drives the vehicles to some different locations around the map. I use the 2020.06 version of LGSVL, Borregas Ave map, Ubuntu 18.04 and Python 3.6.9.

The behaviour can even rarely  get incosistent between the simulator runs without changing the source. You should be able to reproduce by running API only simulator and the video_full_test.py python file. Here in my last runs vehicle number 9 (in code) is not going to its destination although on_waypoint_reach says otherwise. You can see the correct waypoints by running that vehicle only (by changing TRACKED_VEHICLES = [2, 3, 6, 9] variable to TRACKED_VEHICLES = [9] in source code). You can also change SPEED_MULTIPLIER_X variables to see how it causes a vehicle to not reach its intended destination. 

I suspect this can be a bug related with NPC agents kind of sharing same waypoint reached callbacks in backend, causing them to terminate early when one NPC has reached its destination.

[lgsvl_waypoint_İssue.zip](https://github.com/lgsvl/simulator/files/5711487/lgsvl_waypoint_Issue.zip)

vatanaksoytezer, 2021-01-14:
I managed to get this working by calculating timestamps of the NPCs that I would like to move and assigning these timestamps accordingly. As far as I can understand the NPCs calculate their own timestamp if not given any and it causes sim time to mess up if you have an externally controlled vehicle (such as an ego vehicle). Still there are some slight stopping kind of behavior in every callback (which is expected I believe) and time kind of runs slow (which may also be because of a low resource PC of mine can't handle callbacks fast). I was experimenting with moving the ego and NPCs together and as soon as NPCs are done with their waypoints, ego gets faster (not in terms of speed but in terms of flow of time). I will try to send a video of the behavior soon and see if it helps identifying the issue.

EricBoiseLGSVL, 2021-01-15:
@hadiTab is this expected behavior from python api?

hadiTab, 2021-01-15:
@vatanaksoytezer Stopping at every callback is expected. Every API call will cause the simulator to pause for a frame.
The waypoint mode can be inaccurate if a timestamp is not entered and internal calculations are used. This is especially an issue when the vehicle stops or travels at very low speeds. For such applications I'd advise using timestamps which are more accurate.