lgsvl/simulator #1159: how to get smooth curve lane? by the increasing the number of way-point? it doesn't work.

Addi2020, 2020-12-03:
I updated the lanes annotation by increasing the number of way points to get smooth curve (fig1), but when i export the HD map and upload it to dreamview, it is the same as before(fig2), the curve is not smooth. 
fig 1
[image]

fig 2
[image]

luqiang21, 2020-12-03:
@Addi2020 I believe Apollo map format doesn't have the math equations/parameters to describe a curve. As you can see in their demo map https://github.com/ApolloAuto/apollo/blob/master/modules/map/data/demo/base_map.txt, only discrete points are provided for the center, left boundary, right boundary. If you want a smooth curve, the only way might be continue increasing the number of points.

Addi2020, 2020-12-06:
@luqiang21 , I understand , As you can see I already increase number of points in Unity and the curve shown in unity is very smooth, but when I export HD map and upload it to dream view the curve is always same, did not increased number of way points.

luqiang21, 2020-12-07:
@Addi2020 Sorry I misunderstood your problem. As far as I know, the exporter should export the same number of points in your map in Unity. When you export a new HD map after changing number of points in Unity, did you regenerate other files for HD map in apollo?

Addi2020, 2020-12-09:
@luqiang21 , I increase number of points to smoother the curve but after saving the file and export HD map, export HD map and upload it to dream view the curve is always same, did not increased number of way points. there is some problem in unity.

luqiang21, 2020-12-09:
@Addi2020 Did you regenerate files like `routing_map.bin`, `sim_map.bin`, etc. after you update `base_map.bin` in apollo?

Addi2020, 2020-12-10:
@luqiang21  No ,I did not update, i think it (routing_map.bin, sim_map.bin) will update auto. 
could you please tell me how to update these files?

luqiang21, 2020-12-10:
@Addi2020 If you are using our apollo-5.0 fork, you can run `generate_map.sh your_map_folder_name` inside apollo docker to generate those files. If you are using apollo official repo, you can copy `generate_map.sh` file from our fork and do same thing.

closed 2021-07-29 by Addi2020