lgsvl/simulator #1161: Ubuntu 18.04 Simulator stuck at localhost web loading

Jiaqing-Cao, 2020-12-03:
Hi, I am using 2020.06 Simulator on Ubuntu 18.04, the Simulator works fine on Windows. 

However, when I ran it on Ubuntu, it will stuck at localhost web loading or keep going between loading and account,lgsvlsimulator.com.

It look like the image below:
[image]

I have tried other localhost port like 8081 and 8800, also tired to use Ctrl+r to reload page. it still look like above.
[Player.log](https://github.com/lgsvl/simulator/files/5634422/Player.log)
Also attach the Player.log.
Could you please tell me which part is wrong?

EricBoiseLGSVL, 2020-12-07:
@zelenkovsky  Looks like this is the issue
```
WebException: Error: NameResolutionFailure
```
Any ideas on how to resolve this?

Jiaqing-Cao, 2020-12-11:
@zelenkovsky

Could you please give me some help on this issue ?

Thank you!

EricBoiseLGSVL, 2020-12-11:
@Jiaqing-Cao Have you looked at this [issue](https://github.com/lgsvl/simulator/issues/859)?

Jiaqing-Cao, 2020-12-11:
Hi @EricBoiseLGSVL ,

I have looked at the issue you shared, and I tried all the methods David provides, it did not work on my PC.

I also tried GuoliangPeng's "Turn off automatic proxy setting and manual proxy setting", then it will stuck at the LG account sign in page like:

[image]

This way I am not able to even get to page localhost.

Do you have any ideas about that?

EricBoiseLGSVL, 2020-12-11:
@thisisdavidfrank do you have any idea how to further troubleshoot this?

EricBoiseLGSVL, 2020-12-11:
@Jiaqing-Cao did you clear cookies yet?

Jiaqing-Cao, 2020-12-13:
@EricBoiseLGSVL ， I tried to clear cookies, it still did not work.

thisisdavidfrank, 2020-12-14:
Hey @Jiaqing-Cao, so looks like if you remove your Proxy settings, the site is stuck at the Accounts page, correct? So, can you sent the Network Logs when you are on this page and try to Sign In?

It looks like either the Sign In API is not working, or it is working but is not able to set Cookies. So, your Network Logs might help us debug this problem.

Jiaqing-Cao, 2020-12-14:
Hi @thisisdavidfrank ,

Yes，if I remove my Proxy settings, the site is stuck at the Accounts page.

And I get my Network Logs, It looks like that github do not support HAR file type, I will just paste it here:

**{
  "log": {
    "version": "1.2",
    "creator": {
      "name": "WebInspector",
      "version": "537.36"
    },
    "pages": [],
    "entries": []
  }
}**

It only has those lines above, no other logs.

Does that means the Sign In API is not working?

thisisdavidfrank, 2020-12-15:
Hi @Jiaqing-Cao,

Looks like the complete network logs were not captured due to the browser refreshing. 

Can you do the following steps, please?
1. Open a Chrome browser and then Open Developer Tools, and click on the Network tab.
2. Check the two options: Preserve Logs and Disable Cache (Screenshot attached).
3. Close the browser.
4. Now, launch the localhost from the Simulator.
5. After you see the same problem, please download the Network Logs.
6. You can change the extension of the file to .txt and attach it here.

[image]

Thank you.

Jiaqing-Cao, 2020-12-15:
Hi @thisisdavidfrank , I followed your step and get:
[account.lgsvlsimulator.com.txt](https://github.com/lgsvl/simulator/files/5699183/account.lgsvlsimulator.com.txt)
[image]
The first log is I stuck at sign in and get.

[account.lgsvlsimulator.com1.txt](https://github.com/lgsvl/simulator/files/5699189/account.lgsvlsimulator.com1.txt)
[image]
The second log is after I stuck, I used Ctrl+R to reload and get.

Do those logs help you debug?

thisisdavidfrank, 2020-12-15:
Hi @Jiaqing-Cao, thank you for the logs. Unfortunately, I do not see the 'login' API going out in your logs. Can you capture the logs when you also click on the Sign In button, please? That's the only way that I can compare the token being returned by the 'login' API with the token that you have during the redirect.
Thank you.

Jiaqing-Cao, 2020-12-16:
Hi @thisisdavidfrank ,

[account.lgsvlsimulator.com2.txt](https://github.com/lgsvl/simulator/files/5699401/account.lgsvlsimulator.com2.txt)
[image]

I think this log capture the Sign in click.
Will this log help you ?

thisisdavidfrank, 2020-12-22:
Hi @Jiaqing-Cao, thank you again for your log. I've examined it extensively and it seems that the problem is that your API calls for 'token' and 'sign-in' are not reaching the server. If you notice the response size, starting from the 'token' API, all the responses have 0B sizes. The most likely reason for this is that the Linux machine has a restriction on the number of concurrent outgoing TCP connections, and this is causing the connections to fail. Can you increase the number of outgoing TCP connections, and try again?

Note: For security reasons, please **delete** the log attachment in your previous response.

Jiaqing-Cao, 2020-12-22:
Hi @thisisdavidfrank , thanks for your reply.

I followed [this link](https://stackoverflow.com/questions/410616/increasing-the-maximum-number-of-tcp-ip-connections-in-linux)
The top answer to increase the number of outgoing TCP connections.
However, still that problem is not fixed.

And I delete the log attachment in my previous response.

martins-mozeiko, 2020-12-22:
@thisisdavidfrank fyi, you cannot delete attachments in your comments. They will be always available in history ("edited" link).

@Jiaqing-Cao I suggest you to change your password (`139Fal...`), as that is visible in plain text in the attachment you posted. And in case you are using this password in other services, change it there ASAP.

Jiaqing-Cao, 2020-12-22:
@martins-mozeiko , okay, thanks for that.

zelenkovsky, 2020-12-22:
I apologize for inconvenience. Our server went down around 4 hours ago. But now problem is solved. Please try to login again.

**NO NEED TO CLEAR COOKIES**

[image]

closed 2020-12-22 by zelenkovsky