lgsvl/simulator #1162: Issue using 3D Ground Sensor with ROS2

C-500788, 2020-12-03:
Hello, I am using version 2020.6 of the LGSVL simulator on Ubuntu 18.04 for ROS2 I use the eloquent distribution. I am trying to get the position of vehicles generated through the command sim.add_random_agents.
I have set up the ros2-bridge and all the other sensors work. But when I try to load a simulation with a vehicle that has a 3D Ground Truth sensor,  I get the following error in the local host:

Invalid: ROS2 does not allow part topic name start with digit - '/simulator/ground_truth/3d_detections'

hadiTab, 2020-12-06:
@C-500788 the topic name (or any part following the / symbol) cannot start with a number in ROS2. Just change the topic name to somthing else like:  '/simulator/ground_truth/detections_3d' and it should work.

closed 2020-12-16 by EricBoiseLGSVL

arjunskumar, 2021-03-16:
`/simulator/ground_truth/2d_detections` is also having the same issues

C-500788, 2021-03-18:
I think that the ground truth messages are only implemented in ROS not in ROS2

arjunskumar, 2021-03-19:
@C-500788 Is  there any workaround  to obtain ground truth msgs in ROS2?

EricBoiseLGSVL, 2021-03-19:
Yes, add the conversions to the source code for ROS2.  See #1085