lgsvl/simulator #1164: Question about import point Cloud?

daohu527, 2020-12-05:
I know that lgsvl implements the function of importing point clouds as an environment [docs](https://www.lgsvlsimulator.com/docs/pointcloud-import/). Then I have a few question about it.

1. I watched the demo video. It seems that the ground is very flat. When I use 3D modeling to generate the ground, it cannot generate such a smooth ground. Have you done any special treatment on generate the ground?

2. After the environment is generated, if the environment is rendered by a point cloud. how to generate a simulated point cloud of lidar in the simulator? If it through the rays, we cannot guarantee that the ray just falls on the point, not the gap between the two points.

EricBoiseLGSVL, 2020-12-09:
@daohu527 
1) We treat our point cloud data before importing (raw data has a large amout of noise that needs filtered).  We have a beta premium service called Digital Twin that improves raw point cloud data.  This should be available later next year.  "3D Modeling"?  Are you bring the point cloud into a 3D modeling program?
2) Lidar gets the point cloud projection from the points and merge into the z buffer (depth) map.  Gaps are filled by interpolating between the points.
 We use a similar technique as [this](https://www.youtube.com/watch?v=KUQfsZo_iu0).

daohu527, 2020-12-10:
Thank you for your answer, what is the actual result of the lidar point cloud? I saw uber's paper similar to this idea. 

“LiDARsim: Realistic LiDAR Simulation by Leveraging the Real World” [arXiv](https://arxiv.org/abs/2006.09348)

EricBoiseLGSVL, 2020-12-10:
We simplify and extract key data that converts to better point clouds.  @daviduhm can let you know more about this very cool tech