lgsvl/simulator #1165: How to  Use the Apollo HD Map  exported by LGSVL?

Gump0116, 2020-12-07:
It may be a little bit inappropriate to ask here.I now get a base_map.bin here.I want to load it into Apollo DreamView，I have already seen the tutorial that Apollo offerd,but I think it's a bit unspecific.Has anyone done a similar job？

ntutangyun, 2020-12-07:
create a folder under /apollo/modules/map/data and put your map files (xxx.bin) under it. 

restart dreamview 

then you will see the map option the drop down menu in Dreamview.

Gump0116, 2020-12-07:
> create a folder under /apollo/modules/map/data and put your map files (xxx.bin) under it.
> 
> restart dreamview
> 
> then you will see the map option the drop down menu in Dreamview.

Thank you.I find it.If i want to use it with LGSVL,and I want to use the routing Editor Module in DreamView.What other steps should I do

ntutangyun, 2020-12-07:
Once the map is selected and the localization module is working. you will see your vehicle spawned on the map in Dreamview, then you could use the routing editor in Dreamview to submit routing request (routing module running)

Gump0116, 2020-12-07:
> Once the map is selected and the localization module is working. you will see your vehicle spawned on the map in Dreamview, then you could use the routing editor in Dreamview to submit routing request (routing module running)
I only have base_map.bin.Is it OK?

ntutangyun, 2020-12-07:
you need three files as shown [here](https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/map/data/borregas_ave).

Gump0116, 2020-12-07:
> you need three files as shown [here](https://github.com/lgsvl/apollo-5.0/tree/simulator/modules/map/data/borregas_ave).

Now i have two questions.
1.I tried to generate routing_map.bin and sim_map.bin,the problem on the right side of the picture showed.I will try to download libglog.so.0. 
2.What's the meaning of the left side of the picture?Do I need to generate these files？

Gump0116, 2020-12-07:
[need.zip](https://github.com/lgsvl/simulator/files/5650701/need.zip)

This is the picture,Sorry,I am new to using github

lemketron, 2020-12-07:
> 1.I tried to generate routing_map.bin and sim_map.bin,the problem on the right side of the picture showed.I will try to download libglog.so.0.

I believe the problem is that you need to be inside the apollo docker to generate the map files.
Type this first: `docker/scripts/dev_into.sh` 
(assuming you have already done `docker/scripts/dev_start.sh`)

Gump0116, 2020-12-08:
> I believe the problem is that you need to be inside the apollo docker to generate the map files.
> Type this first: `docker/scripts/dev_into.sh`
> (assuming you have already done [...]

Thank you,I made it inside the apollo docker!

Gump0116, 2020-12-08:
Now i know how to use it,With base_map.bin and routing_map.bin and sim_map.bin,you can run routing Editor Module!Thank your help!@lemketron  @ntutangyun.

closed 2020-12-16 by EricBoiseLGSVL