lgsvl/simulator #1166: LGSVL Simulator (linux64-2020.06) won't start :(

ironsuperdev, 2020-12-07:
Hi, 

I downloaded the newst version of LGSVL lgsvlsimulator-linux64-2020.06 under ubuntu 18.04, trying to launch the program, but the window appears and then disappears after seconds. 

Then I checked my NV driver version and the log as follows

I tried all NV driver packages available on my computer, none of them is working.

[https://pasteboard.co/JDLAeNA.png](https://pasteboard.co/JDLAeNA.png)

My NV Graphic Card is GeForce RTX 2070. Does anyone know how to resolve this?

Thank u indeed for any advice. 

```
Mono path[0] = '/media/song/Transfer/Work/AD/LGSVL/lgsvlsimulator-linux64-2020.06/simulator_Data/Managed'
Mono config path = '/media/song/Transfer/Work/AD/LGSVL/lgsvlsimulator-linux64-2020.06/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 'HP E243i 24"': 1920x1200 (primary device).
Loading player data from /media/song/Transfer/Work/AD/LGSVL/lgsvlsimulator-linux64-2020.06/simulator_Data/data.unity3d
Desktop is 1920 x 1200 @ 60 Hz
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
Vulkan detection: 2
Initialize engine version: 2019.3.15f1 (59ff3e03856d)
[Subsystems] Discovering subsystems at path /media/song/Transfer/Work/AD/LGSVL/lgsvlsimulator-linux64-2020.06/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=81
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[... 129 lines cut ...]
```
~$ nvidia-smi
Mon Dec  7 12:10:22 2020       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 440.95.01    Driver Version: 440.95.01    CUDA Version: 10.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce RTX 2070    Off  | 00000000:01:00.0  On |                  N/A |
| 43%   28C    P8    11W / 175W |    376MiB /  7979MiB |      2%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1210      G   /usr/lib/xorg/Xorg                            18MiB |
|    0      1316      G   /usr/bin/gnome-shell                          11MiB |
|    0      9317      G   /usr/lib/xorg/Xorg                           158MiB |
|    0      9435      G   /usr/bin/gnome-shell                         103MiB |
|    0      9956      G   ...AAAAAAAAAAAIAAAAAAAAAA== --shared-files    79MiB |
+-----------------------------------------------------------------------------+

```

ironsuperdev, 2020-12-07:
https://github.com/lgsvl/simulator/issues/879

hi, I solved the problem by check the link above. The vulkan driver is falsely installed.

closed 2020-12-07 by ironsuperdev

lemketron, 2020-12-07:
> hi, I solved the problem by check the link above. The vulkan driver is falsely installed.

Good catch. Yes, I noticed the Intel GPU mentioned as the vulkan renderer and was going to suggest the same fix.