lgsvl/simulator #1168: calculate the heading angle in the local coordinate frame

manonechu, 2020-12-08:
Hi,

In reality, a GPS can return lateral and longitudinal coordinates, as well as velocities in the north and east directions. Based on these velocities, it is easy to calculate the heading angle in the local coordinate frame (e.g. north is Y axis, east is X axis).

I would like to know which sensor can return the velocities in the north and east directions in the simulator? Or is there a compass sensor available to get the angle to the north direction?

Thank you very much for your help.

EricBoiseLGSVL, 2020-12-10:
@hadiTab do we have a way with current sensors to get the angle to the north direction of maps?

hadiTab, 2020-12-11:
@manonechu this depends on which bridge you are using. You can use any message that has orientation to figure out the direction. The imu message includes orientation as a quaternion in ROS and ROS2.

EricBoiseLGSVL, 2020-12-11:
Thanks @hadiTab

manonechu, 2020-12-11:
@hadiTab Thank you very much for your help. I just tried using IMU in ROS2 and it works.
@EricBoiseLGSVL Thank you very much for your help!

closed 2020-12-11 by EricBoiseLGSVL