lgsvl/simulator #1169: Unity version 3.15, LGSVL version 2020.06, it is not working

iamshane, 2020-12-09:
[image]
please check the picture here are many errors, even the asset data base is missing, I am wondering if the version for LGSVL or Unity is not working, please give me feedback, thank you very much.

ntutangyun, 2020-12-09:
is it because you didn't install `git lfs` when cloning the project ?

iamshane, 2020-12-10:
after I setup node js and git lfs, it is still error on Unity3D when I try to run the simulator,can you 
please give the reply? thank you.
[image]

EricBoiseLGSVL, 2020-12-10:
Again, you need to git lfs [here](https://git-lfs.github.com/)
git lfs fetch will work since you cloned already, see [here](https://github.com/git-lfs/git-lfs/issues/325)