lgsvl/simulator #1170: Module malfunction in modular testing with apollo

Yimags, 2020-12-09:
Hello all,

I ran the modular test with apollo and came with the problem that all module except for localization and chasis did not work at all (no perception, prediction, 
planning, control etc.). I followed the instructions and built apollo again and again step by step. But it still shows the same result. So I was wondering if you could help to localize the problem. Thanks in advance!
[image]
[image]

Yimags, 2020-12-09:
And here is the log recorded.
[image]

luqiang21, 2020-12-09:
@Yimags It looks like you didn't add modular testing sensors. You need to add them in your sensor configuration.

Yimags, 2020-12-09:
> @Yimags It looks like you didn't add modular testing sensors. You need to add them in your sensor configuration.

Ahh exactly, I have changed it to my personal settings. Thx!

I am also wondering to what extent I could customize the sensor settings without causeing the malfunction of Apollo? Or how could I modify Apollo if I want to change sensors in lgsvl?

luqiang21, 2020-12-10:
@Yimags You just need to list which sensors are needed by those apollo modules, make sure those sensors are added in your sensor configuration. If an Apollo module is not working, you can first check the existence of the corresponding sensor. Then check apollo module's log file.

What do you mean?
> Or how could I modify Apollo if I want to change sensors in lgsvl?

Yimags, 2020-12-10:
> @Yimags You just need to list which sensors are needed by those apollo modules, make sure those sensors are added in your sensor configuration. If an Apollo module is not working, you can first [...]

I meant whether newly added senors (apart from those of modular test) will influence influence the functionality or not. If it does, how could I modify apollo code to make new sensor work well with apollo?

luqiang21, 2020-12-11:
@Yimags It should depend on the topic name of the sensor you add. If they are different from what Apollo is using, then it won't influence.

Yimags, 2020-12-11:
> @Yimags It should depend on the topic name of the sensor you add. If they are different from what Apollo is using, then it won't influence.

Got it. Thank you for your help！

closed 2020-12-11 by Yimags

roshambo919, 2021-10-07:
nvm. user error. for future users, configuration is located in the vehicle library on wise.svlsimulator.com