lgsvl/simulator #1171: Simulator does not bind to 8181

ar4development, 2020-12-11:
**Simulator version:** 2020.06
**OS:** Ubuntu 20.04
**Config:**
```
hostname: "localhost"
port: 8080
headless: false
read_only: false
api_hostname: "localhost"
api_port: 8181
```

**OBSERVED:**  After the simulator has started, I'm doing `netstat -tulpn` and there I can see that the process is listening on port `8080` and it also holds UDP port `9999`. However port `8181` is not listed there. So everything is working but the API. No issues with trying to obtain `8181` port found neither in syslog nor in ufw log.
**EXPECTED:** Port `8181` is also listened by the simulator

**Log:**
```
Mono path[0] = '/masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/Managed'
Mono config path = '/masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 'VA2407 Series 24"': 1920x1080 (primary device).
Loading player data from /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/data.unity3d
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 2
Initialize engine version: 2019.3.15f1 (59ff3e03856d)
[Subsystems] Discovering subsystems at path /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=18
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_wayland_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=111
[... 148 lines cut ...]

Fallback handler could not load library /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/data-0x447f670.so
Loaded 3 NPCs behaviours and 6 NPC models in 00:00:00.1870847 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

Fallback handler could not load library /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/libKernel32
Fallback handler could not load library /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/libKernel32.so
Fallback handler could not load library /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/Kernel32
Fallback handler could not load library /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/libKernel32
Fallback handler could not load library /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/libKernel32.so
Fallback handler could not load library /masked_path/lgsvlsimulator-linux64-2020.06/simulator_Data/Mono/libKernel32
Current Database Version: 3 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

Final Database Version: 3 
(Filename: ./Runtime/Export/Debug/Debug.bindings.h Line: 35)

Setting up 6 worker threads for Enlighten.
  Thread -> id: 7fe0b082b700 -> priority: 1 
  Thread -> id: 7fe05bfff700 -> priority: 1 
  Thread -> id: 7fe05b7fe700 -> priority: 1 
  Thread -> id: 7fe05affd700 -> priority: 1 
  Thread -> id: 7fe05a7fc700 -> priority: 1 
  Thread -> id: 7fe059ffb700 -> priority: 1 
```

EricBoiseLGSVL, 2020-12-12:
@ar4development See if [this](https://github.com/lgsvl/simulator/issues/868) issue helps.

martins-mozeiko, 2020-12-12:
Have you started simulation (from web interface) that enables API access? Simulator listens on API port only when simulation with API mode enabled is running.

ar4development, 2020-12-12:
@EricBoiseLGSVL no that one is a different issue. That issue describes the case when `8181` port is listened and listed with `netstat`. In my case I do not have that port listening so that no matter which IP address I'm using.

ar4development, 2020-12-12:
@martins-mozeiko yes that helped. Thank you! Is there a way to enable API right on start up without manual actions in Web UI?

martins-mozeiko, 2020-12-16:
You cannot enable API simulation right on start up. But you can use same command as website does - send simulator to start API simulation. Check the network inspector in chrome of firefox to see what JSON request it sends to port 8080 to do that. This is really "undocumented" thing and probably will change in next versions. Maybe @EricBoiseLGSVL can give you more information on what is planned next to simplify this.

EricBoiseLGSVL, 2020-12-17:
@ar4development Yes, @martins-mozeiko is correct, we only support WebUI access but you can make changes to the source.  We have plans to make CL/CI pipeline better is future releases.