lgsvl/simulator #1172: When or Where can I get the 2020-08-31 simulator

wisdom-bob, 2020-12-12:
Hello, I get the message from Documentation, I find the simulator(in unity) has been updated to 2020-08-31. However, I can't find the link to download the latest simulator. 
So.... Could you please tell me where I can get this ?

ntutangyun, 2020-12-14:
could you leave the link to the documentation or the message ?

EricBoiseLGSVL, 2020-12-15:
@wisdom-bob The latest public release is [here](https://github.com/lgsvl/simulator/tree/release-2020.06).  With Unity 2019.3.15f1.  What are you referencing?

wisdom-bob, 2020-12-16:
> @wisdom-bob The latest public release is [here](https://github.com/lgsvl/simulator/tree/release-2020.06). With Unity 2019.3.15f1. What are you referencing?

thanks a lot！by the way, when the curve application will be ok rather than building with linestrings?