lgsvl/simulator #1174: Error of 'No module named '_cyber_wrapper'' when trying to get data from cyberRT with pythonAPI

Yimags, 2020-12-14:
Hello all,

I am now running lgsvl with apollo and trying to get data from cyberRT with pythonAPI. However I come across an error when running [listener.py](https://github.com/ApolloAuto/apollo/blob/master/cyber/python/cyber_py3/examples/listener.py). Any suggestion or solution will be appreciated. Thx in advance!
[image]

ntutangyun, 2020-12-14:
you should probably ask this question at [Apollo's repo](https://github.com/ApolloAuto/apollo/issues)

luqiang21, 2020-12-14:
@Yimags Are you running listener.py inside apollo docker?

Yimags, 2020-12-14:
> @Yimags Are you running listener.py inside apollo docker?

Not quite. I tried it outside the docker. Is it necessary to run the cyber python API inside the docker?

And now I run it inside the docker but another missing module occurs:

[image]

Yimags, 2020-12-29:
The python files in apollo are managed by bazel. So instead of running it directly, running `bazel run //cyber_py3/examples:listener` should work.
For [details](https://github.com/storypku/CyberRT/tree/master/cyber_py3).

closed 2020-12-29 by Yimags