lgsvl/simulator #1176: Git clone fails x509: certificate signed by unknown authority

johschmitz, 2020-12-15:
For me the git clone operation fails with the following error:

```
$ git clone --recursive https://github.com/lgsvl/simulator.git lg_svl_simulator
Cloning into 'lg_svl_simulator'...
remote: Enumerating objects: 1038, done.
remote: Counting objects: 100% (1038/1038), done.
remote: Compressing objects: 100% (513/513), done.
remote: Total 22019 (delta 623), reused 687 (delta 525), pack-reused 20981
Receiving objects: 100% (22019/22019), 11.33 MiB | 2.80 MiB/s, done.
Resolving deltas: 100% (16441/16441), done.
Downloading Assets/Editor/MapUI/MapUIBoundryCurb.png (3.4 KB)
Error downloading object: Assets/Editor/MapUI/MapUIBoundryCurb.png (83864de): Smudge error: Error downloading Assets/Editor/MapUI/MapUIBoundryCurb.png (83864de4660c9b512bd11e01ada86cd2de7c9335b2fe8bab5e28d0b8c9f1ebf3): batch response: Post "https://github.com/lgsvl/simulator.git/info/lfs/objects/batch": x509: certificate signed by unknown authority

Errors logged to /home/schmitz/work/scm_branches/lg_svl_simulator/.git/lfs/logs/20201215T155147.733657063.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: Assets/Editor/MapUI/MapUIBoundryCurb.png: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry the checkout with 'git checkout -f HEAD'
```
See the git lfs log attached.
[lfs_log.txt](https://github.com/lgsvl/simulator/files/5696658/lfs_log.txt)

Any idea how to fix this?

EricBoiseLGSVL, 2020-12-15:
try git lfs pull

martins-mozeiko, 2020-12-16:
Either your host certificates are corrupted/modified, or your network (your company or your ISP) is doing MITM on https connections.

Check that you can access github domain with openssl:
```
openssl s_client -servername github.com -connect github.com:443
```
In output you should see something like this in the beginning:
```
Certificate chain
 0 s:C = US, ST = California, L = San Francisco, O = "GitHub, Inc.", CN = github.com
   i:C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert SHA2 High Assurance Server CA
 1 s:C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert SHA2 High Assurance Server CA
   i:C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert High Assurance EV Root CA
---
```
and a bit below:
```
---
SSL handshake has read 3507 bytes and written 376 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_128_GCM_SHA256
Server public key is 2048 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
```

johschmitz, 2020-12-16:
@martins-mozeiko, @EricBoiseLGSVL  I can access Github without problems and normal clones and pulls (without LFS) work perfectly fine. This here is the only repository so far that shows this issue.

`git lfs pull` results in:
```
batch response: Post "https://github.com/lgsvl/simulator.git/info/lfs/objects/batch": x509: certificate signed by unknown authority
batch response: Post "https://github.com/lgsvl/simulator.git/info/lfs/objects/batch": x509: certificate signed by unknown authority
batch response: Post "https://github.com/lgsvl/simulator.git/info/lfs/objects/batch": x509: certificate signed by unknown authority
error: failed to fetch some objects from 'https://github.com/lgsvl/simulator.git/info/lfs'
```

I also see the LG SVL Simulator code in the directory on my disk after the clone, just not the LFS hosted parts.
I am going to update the title of this issue accordingly.

EricBoiseLGSVL, 2020-12-16:
@johschmitz  it seems git lfs is having issues with certs,  maybe [this](https://github.com/git-lfs/git-lfs/issues/276#issuecomment-98754786) will help

martins-mozeiko, 2020-12-16:
@johschmitz yes, I understand that your normal git access work, but you need to debug git connection - there's not much we can configure in github repository. I'm pretty sure something is wrong with your certificates or some network appliance capturing/corrupting traffic. Maybe it works for regular domain, but not for domain where git lfs fetches files.

Try running git with extra trace enabled:
```
GIT_CURL_VERBOSE=1 GIT_TRACE=1 git clone ...
```
This will show a lot of information. For example for lfs download parts it shows me that gets LFS files from Amazon S3.
```
12:38:47.159667 trace git-lfs: HTTP: GET https://github-cloud.s3.amazonaws.com/alambic/media/180968610/ac/cc/accc891431e15add828e47fedef563
7daa2e233ceaa709902128e340281ca4ce
```

Can you check that your connections to this domain succeed?
```
openssl s_client -servername github-cloud.s3.amazonaws.com -connect github-cloud.s3.amazonaws.com:443
```

I get following output:
```
Certificate chain
 0 s:C = US, ST = Washington, L = Seattle, O = "Amazon.com, Inc.", CN = *.s3.amazonaws.com
   i:C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert Baltimore CA-2 G2
 1 s:C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert Baltimore CA-2 G2
   i:C = IE, O = Baltimore, OU = CyberTrust, CN = Baltimore CyberTrust Root
---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA
Server Temp Key: ECDH, P-256, 256 bits
---
SSL handshake has read 3399 bytes and written 457 bytes
Verification: OK
---
New, TLSv1.2, Cipher is ECDHE-RSA-AES128-GCM-SHA256
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES128-GCM-SHA256
    Session-ID: E1...3C
    Session-ID-ctx:
    Master-Key: 2B...33
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    Start Time: 1608151411
    Timeout   : 7200 (sec)
    Verify return code: 0 (ok)
    Extended master secret: no
---
```

johschmitz, 2020-12-17:
Here is the verbose output [lg_svl_lfs_log.txt](https://github.com/lgsvl/simulator/files/5710088/lg_svl_lfs_log.txt)
Since this does not happen at home I just would like to be able to pinpoint this to the network side so I can tell the IT department guys exactly what I need. However, I am not even reaching the AWS step it seems

As you suggested I checked the connection to AWS itself and it seems to be working fine. I believe the problem must be somewhere in between.

closed 2021-07-16 by EricBoiseLGSVL