lgsvl/simulator #1178: How to build and upload 3D-Horizontal Traffic light Model?

Addi2020, 2020-12-16:
we are a Chinese company XXX and helping the LGSVL eco system to build more traffic light models". Now the thing is LGSVL has to let us know how to build the model in 3DMAX or solid work etc. As there are multiple layer in the model. We tried to build a model but due the layer differences, the green light does not work.
Please let us know in details about;
1. How to create 3D-Horizontal Traffic Light Model using which software
2. How to upload this model to Unity3D to create HD-map

We would be grateful for your assistance.

EricBoiseLGSVL, 2020-12-16:
@Addi2020 We have updated the traffic signal code to be easier to make changes to the signal meshes.  This will be in the next release in Jan.  The current system offsets the emission texture.  You just need to change the single mesh lookup to a struct or list so you can get each renderer and apply emission and color changes to each light independently.

Here you can see the class with the List.  We order the renderers by the mesh name, SignalLight0 and map it to the index, like so
[image]
Each material color is white and on start applies the color to the material.  We will also have emission textures in this release, e.g. left/right/uturn/X.

If you don't want to wait you can use any modeling software you want.  Maya is recommended but you can also use Blender.  Just be sure to make the model Z forward.  Blender has a different coordinate system.  Just use the model we provide as a guide but make each signal light a separate mesh with it's own material.  You should have one material for the case, and a material for each signal light, named SignalLight0..SignalLightn.

Then you can write a simple prefab editor script or manually swap out the meshes in a scene.  Assets/Resources/Editor/EditorSettings.asset, is a reference to the default traffic signal prefab used during hdmap import.  You can replace this with your prefab.  If you have varying meshes, you will need to edit the importer code or just manually swap out after the initial import.

closed 2021-07-29 by Addi2020

EricBoiseLGSVL, 2021-07-29:
@Addi2020 thanks for closing open issues.  Hope your development is going well.

Addi2020, 2021-07-31:
@EricBoiseLGSVL , I close all the pending issues, in which some of them i find solution and some of them I didn't, which  i assume that cannot be solve/deal by Unity Hub. e.g I assume the issue #1422 and #1159 cant be solve by Unity Hub. I temporally close all pending issues, if i need it then I may reopen it or should I leave it open the unsolved issues?

EricBoiseLGSVL, 2021-08-02:
Thanks and yes, please open if you have more information but I don't think they will be solved soon.  One is an issue with apollo and the other looks like it is very custom.  Let's leave them closed until we can work on annotation internally.  We hope to improve it in the next few releases.