lgsvl/simulator #1179: Start the Autoware.Auto containers with NVIDIA setup stuck

qiongwu1003, 2020-12-17:
Hi 
LGSVL team

I built the environment of Autoware.Auto with LGSVL Simulator followed [here](https://www.lgsvlsimulator.com/docs/autoware-auto-instructions/).
When I step into "Start the Autoware.Auto containers with NVIDIA setup:", I am stuck on the "Creating volume container for nvidia/cuda:latest". Could someone give me some advice?
[image]

All things went well except when I first started ade.
When I first started I met the "token" event as below, and I created myself a token. But I met the problem above.
[image]

I think it might be the "token" problem,  so I delete the ade(I am not sure if deleting it directly is correct) and reinstall it. This time I did not meet the "token" event. But I still met the problem above.

Moreover, my Nvidia driver message is as below. And I automatically install it using the PPA repository.
[image]

EricBoiseLGSVL, 2020-12-17:
@qiongwu1003 I am concerned you are using a 1060 GPU, this might have issues running simulator.

qiongwu1003, 2020-12-18:
@EricBoiseLGSVL 
Thanks for your reply.
I changed a newer 2070 GPU, and still have the same issue. 
[image]

qiongwu1003, 2020-12-18:
@EricBoiseLGSVL 
I turned off my network and tried again. It needs "username" and "token", once I enter them it gives out bellow. And I do not think the "username" and "token" are correct since I created them myself.
[image]

Have any ideas according to this output?

lemketron, 2020-12-18:
> When I first started I met the "token" event as below, and I created myself a token. But I met the problem above.

Is "mcw" your gitlab username? Did you create a personal access token on the page shown in the screen shot (https://gitlab.com/-/profile/personal_access_tokens)?

qiongwu1003, 2020-12-18:
@lemketron 

>Is "mcw" your gitlab username? Did you create a personal access token on the page shown in the screen shot
No, it is my system name. 
Yes, I create a personal access token.
You mean I should enter the gitlab username in the "Username:" location?

hadiTab, 2020-12-18:
@qiongwu1003 it seems like it's asking for username and password to pull the docker images from the gitlab registry. Now, I would have assumed that it wouldn't need credentials to pull the image, but maybe I'm wrong. But the credentials, for whatever reason they are being requested, would be your gitlab credentials.

Overall, this seems like an issue with setting up the ADE containers in AutowareAuto, so it's probably best to try to get help from the Autoware people. Best way to get help is to post a question on answers.ros.org and add a tag for autoware.auto. Josh from AWF is usually pretty quick to respond.

qiongwu1003, 2020-12-22:
@hadiTab 
Thank you very much. I will try to get help from Autoware and close the issue here.

closed 2020-12-22 by qiongwu1003