lgsvl/simulator #1180: HD Map Annotation lacking "Create fake boundary lines" utility

Alejandro-A-R, 2020-12-17:
I am annotating a custom HD map, and I have already annotated the lanes. When I try to export the HD map, an error occurs and, if I am not wrong, it has to do with the map not having boundary lines annotated. In https://www.lgsvlsimulator.com/docs/map-annotation/, it states that HD map annotation tool has, under "Utilities", a button named "Create Lines", which creates fake boundary lines for the lanes which lack them (all in my case). The problem is that I can't see the "Utilities" section nor the "Create Lines" button, the only thing I see is the "Snap All" button. I know I can annotate boundary lines manually, but that will take me several days that I would prefer not spending in creating such lines.
[image]

luqiang21, 2020-12-17:
@Alejandro-A-R Are you using correct branch? The latest code is available in https://github.com/lgsvl/simulator/tree/release-2020.06 not master.

Alejandro-A-R, 2020-12-17:
Thank you, I thought I was using that version, but in fact I was using the master branch. Now I can use the "Create Lines" tool and it seems to work fine.

closed 2020-12-17 by Alejandro-A-R