lgsvl/simulator #1181: Change Camera Mode, Position and Orientation from Python API

vatanaksoytezer, 2020-12-17:
Hello,

I am happily using LGSVL for a while and at some point I realized that in every simulation run I change the camera to Free mode and put it in a higher position so I can see the scene clearly. As much as I could dig the API I couldn't find such behavior, though I would appreciate if LGSVL team has any plans to add this as a feature. I think I should be able to change it via Unity but I won't want to rebuild the simulation every time I need to change the angle or position of the camera, and I would be happy to do it through API since I change scenes and configurations quite frequently and most of the work is somehow automated. I would also be happy to contribute if you can guide me through.

EricBoiseLGSVL, 2020-12-17:
@vatanaksoytezer This is a great idea.  Let me see if we can add this before our next release.

EricBoiseLGSVL, 2021-01-14:
This has been added and will be in the next release @vatanaksoytezer

closed 2021-01-14 by EricBoiseLGSVL

vatanaksoytezer, 2021-01-14:
That is great, thanks for considering and implementing this feature! Any planned date on the new release?

EricBoiseLGSVL, 2021-01-15:
No prob, it's a good feature for users, thanks for letting us know about it.  Should be feb for release. It's big so we have a bunch of testing and bugs to squash :)