lgsvl/simulator #1182: How to use DriveWaypoint timestamp parameter?

vatanaksoytezer, 2020-12-17:
I have been working with DriveWaypoint for a while, but now I need to simulate a timeline and I noticed DriveWaypoint object has a parameter called timestamp. I couldn't find any explanation in the API guide. Could you eloborate on what kind of timestamp should we use and in which type?

vatanaksoytezer, 2020-12-22:
Any news on this issue @hadiTab ? This would be a quite useful feature for me if it is what I think. I can also help with the documentation as well.

vatanaksoytezer, 2021-01-14:
I think I figured this out, and got it working. You guys need any documentation on this @EricBoiseLGSVL? If no need for it, I think I can close the issue.

EricBoiseLGSVL, 2021-01-15:
Let's see if @hadiTab can clarify.  We are putting together the next big public release so any updates to docs are great.  You could post here and we can review.

hadiTab, 2021-01-15:
@vatanaksoytezer sorry I missed this one. The `timestamp` field is the simulation time for when we want the NPC vehicle to have arrived at the waypoint. It's an optional field and will be calculated in the simulator based on velocity if it is unassigned.

I guess this one slipped through and didn't get documented. We will update it. Thanks.