lgsvl/simulator #1183: Issues related to traffic lights?

Addi2020, 2020-12-18:
I am facing may problems while creating traffic lights. Your response is highly appreciated.
1. I changed the height of traffic light by changing Y-axis of Position (Pic-1), but it does not effect in real world (pic-2).
2.  what bound offset and bound scale means (Pic-1)? 
3. I create traffic light with signal type MIX_3_Horizontal , but signal light Mesh is Vertical, so when I upload it to Dreamview (Pic-3), It creates Vertical model of Traffic light rather then horizontal. 

Pic-1
[image]

Pic-2
[image]

Pic-3
[image]

EricBoiseLGSVL, 2020-12-19:
1) If you rotate the mesh Y is no longer height.  Prob -+X
2) Bounding box for the light mesh for the AD stack
3) we only support 3 vertical in the HDmap export.  You just need to edit to be able to add horizontal

Addi2020, 2020-12-19:
@EricBoiseLGSVL , can you please explain your answers?
1. you mean height of traffic light depends on X prob?
2. AD stack?
3. do i need to edit signal mesh script?

EricBoiseLGSVL, 2020-12-21:
1) I mean that X is up down after you rotate the light on the mesh.  Did you export the map and convert to .bin files for apollo after adjusting?  Is the signal in the correct position in the scene?  Are you correctly localizing the video to the scene, scale, position, rotation, camera properties (focal length, etc)

2) BoundScale is the bounding box around the traffic signal.  This is used by the AD stack (Autonomous Driving) Apollo, Autoware, etc.

3) In our exporter we default to Vertical 3 type.  If you want to use a horizontal, you need to edit the exporter to support this.  It is in the HDMap exporter code for the AD stack you use

[image]

Addi2020, 2020-12-22:
@EricBoiseLGSVL  got it. Thanks

closed 2020-12-22 by Addi2020