lgsvl/simulator #1185: CameraInfo message (ROS2)

soldierofhell, 2020-12-18:
Hi, any plans to include [CameraInfo.msg](https://docs.ros.org/en/api/sensor_msgs/html/msg/CameraInfo.html) to camera image output? Regards,

EricBoiseLGSVL, 2020-12-21:
@soldierofhell Not right now but most likely next year.  Right @hadiTab ?

hadiTab, 2020-12-21:
Yeah, it's in the plans but can't give an exact time for it.

Thinsheep, 2021-05-22:
@EricBoiseLGSVL @hadiTab 
Does it mean that  CameraInfo.msg is not sent to Apollo? and Apollo uses the camera configuration with the default values?
If I am correct the image data will be sent to Apollo using the topic
"Topic": "/apollo/sensor/camera/front_6mm/image/compressed"
Writing two data types "CameraInfo.msg" and "Image.msg" to the same topic may not be the right idea?
Do you know which topic ( channel) apollo uses to subscribe for camera info?