lgsvl/simulator #1187: Orientation of NPC vehicles

MartinaB85, 2020-12-21:
Good morning,

we've been adding a new custom map to the lgsvl simulator and when connecting it to Apollo 5.0 the orientation of the spawned NPC vehicles is wrong as you can see in the video in attached. The NPC are correctly moving forward but their orientation and the bounding boxes detected by Apollo are horizontally oriented going in the left direction.

Is it something due to wrong annotation map? How can we fix this?

Thank you very much,

Martina

https://user-images.githubusercontent.com/73334464/102770945-4149d080-4385-11eb-9575-f094bdfd6923.mp4

EricBoiseLGSVL, 2020-12-21:
@MartinaB85 Morning, I think this is an issue with the map not being rotated correctly.  @hadiTab ?

hadiTab, 2020-12-21:
@MartinaB85 as @EricBoiseLGSVL pointed out this is likely an issue with map orientation. The map is expected to be orientated in the Unity editor such that the world z-axis is pointing East. It's common for people to orient it so that Z points North, which will cause issues in orientation measurements.

MartinaB85, 2020-12-23:
Thank you so much, that solved the issue.

I would like to ask you also...I want to create a scenario with traffic cones but after building the asset bundle as explained in https://github.com/lgsvl/TrafficCone, changed the name as it is used in the python API, i.e. "TrafficCone", I keep getting Unknown 'TrafficCone' controllable prefab. I have tried rebuilding several times but it does not work. 
Attached are some screenshot of what it has been created in the controllable folder and the name it is called in the code.

Hopefully you can help me.

Thank you very much,

Martina
[image]
[image]
[image]

EricBoiseLGSVL, 2020-12-28:
You have not built the controllable correctly if the name of the file is TrafficCone.pdb.  Did you have errors when you built the controllable in the Unity Editor Simulator -> Build... menu?
It should look like this...
[image]

Also, if you want to create a new controllable, don't name it exactly the same as the one you already have loaded.  Please Prepend with MyTrafficCone or something.  I have a feeling there are class name conflicts here.

MartinaB85, 2021-01-12:
Thank you. The problem apparently was the name I was using in python.
Thank you very much.! :)

closed 2021-01-12 by EricBoiseLGSVL