lgsvl/simulator #1188: Unity issue with Web UI

tallid, 2020-12-21:
Hello ,

I am trying to connect the simulator to Unity but I am having some issue. Any help would be greatly appreciated.
While following the instructions on this link: https://www.lgsvlsimulator.com/docs/build-instructions/

Version is as follows:
- Ubuntu 20.04
- Unity 2019.3.15f1.
- LGSVL  2020.06

The first issue I encountered was in Step 2: I was unable to see Mono support for Linux. Thus I selected the following:
Android Build Support
Linux Build Support(IL2CPP)
Mac Build Support (Mono)
WebGL Build Support
Windows Build Support (Mono)

The next and main Issue I am facing is in step 12 and step 13. I have attached the corresponding Images.
I am unable to Build Web UI. Additionally, on the top, I don't seem to have the drop-down option for 'internal'

Any help would be greatly appreciated. 
[image]

EricBoiseLGSVL, 2020-12-21:
@tallid if you are on Linux, just install Windows Build Support (Mono).  Don't install the other modules.
in the WebUI folder in the repo
1) npm install
2) npm run pack
Internal?

ntutangyun, 2020-12-29:
There seems to be an error regarding web UI building. 

you could try in terminal:

```bash
cd simulator/WebUI/
npm install
npm run pack
```

and see if you could build the web UI manually.

ZhenshengLee, 2021-03-24:
`npm run pack` did not work, `npm pack` just works.

closed 2021-07-16 by EricBoiseLGSVL