lgsvl/simulator #1191: Question about improve Export Opendrive map

wisdom-bob, 2020-12-27:
Hello, in the use of LGSVL2020.06, I found that at present the Opendrive format map exported is much better than the original, the road reference line meet continuous conductivity, but for road boundry, line corss and smooth problem is not covered so that the road is not suitable for pathplanning. By checking, I found that parampoly3 curve does not support the migration boundry continuous differentiable nature (parampoly3 is a kind of variable curvature of curve, different from poly3 curve, can represent multiple poly3 curves by one curve.). Can road offset be actually considered through Arc + Lanesection offset (Poly3)?
it's hard to calculate a spiral + arc like what the opendrive official did. Maybe it's a way to solve the problem.
Here's a sample road map data I built from the opendrive official map tool.
<road name="" length="72.94334677170826" id="1" junction="-1">
		<link/>
		<type s="0" type="town">
			<speed max="25" unit="km/h"/>
		</type>
		<planView>
			<geometry s="0" x="123.7922381789" y="416.7100916538" hdg="1.403105275634046" length="32.7310380396958">
				<arc curvature="0.013578249187502063"/>
			</geometry>
			<geometry s="32.7310380396958" x="122.0231376091" y="449.124178799" hdg="1.8475354663026446" length="32.855809357171914">
				<arc curvature="0.028300048480055007"/>
			</geometry>
			<geometry s="65.58684739686771" x="100.6197895617" y="472.4872417124" hdg="2.7773564639620547" length="7.35649937484056">
				<arc curvature="0.016670882248861443"/>
			</geometry>
		</planView>
		<elevationProfile/>
		<lateralProfile/>
		<lanes>
			<laneOffset s="0" a="0.16542623475743135" b="0.008510984125241318" c="-0.0010100450256622844" d="0.0001459242588305436"/>
			<laneOffset s="11.236027983776168" a="0.3405372947415927" b="0.03785075151875206" c="-0.0038716764700997665" d="0.000046610345219742735"/>
			<laneOffset s="27.357285525715888" a="0.1397994442012023" b="-0.04546238325269295" c="0.004791284602833267" d="-0.00013015107314527527"/>
			<laneOffset s="42.53874232541875" a="0.09849853627729253" b="0.019822233941534845" c="0.003501686026777744" d="-0.0002850103818196795"/>
			<laneOffset s="49.894520539710975" a="0.3203390104438999" b="0.058406333165819085" c="-0.008733096519224923" d="0.0005582912278693175"/>
			<laneOffset s="60.19261003972008" a="0.6053826073575818" b="0.051538227078340354" c="-0.010559730778152136" d="0.0007648611269168752"/>
			<laneOffset s="71.10422192799813" a="0.9041579601534904" b="0.09856013650845177" c="0.009657187638338184" d="-0.003599302281688902"/>
			<laneSection s="0">
				<left>
					<lane id="1" type="driving">
						<link>
							<predecessor id="1"/>
							<successor id="1"/>
						</link>
						<width sOffset="0" a="3" b="0" c="0" d="0"/>
						<roadMark sOffset="0" type="solid" width="0.13" color="white" laneChange="none"/>
						<speed sOffset="0" max="25" unit="km/h"/>
					</lane>
				</left>
				<center>
					<lane id="0" type="driving">
						<link/>
						<roadMark sOffset="0" type="solid" width="0.13" color="white" laneChange="none"/>
					</lane>
				</center>
				<right>
					<lane id="-1" type="bidirectional">
						<link>
							<predecessor id="-1"/>
							<successor id="-1"/>
						</link>
						<width sOffset="0" a="3" b="0" c="0" d="0"/>
						<roadMark sOffset="0" type="none" width="0.13" color="white" laneChange="both"/>
					</lane>
				</right>
			</laneSection>
		</lanes>
		<objects/>
		<signals/>
	</road>

luqiang21, 2020-12-28:
Hi @wisdom-bob what do you mean by `Can road offset be actually considered through Arc + Lanesection offset (Poly3)?` I am not  very clear about your question. Can you clarify?

wisdom-bob, 2020-12-29:
> Hi @wisdom-bob what do you mean by `Can road offset be actually considered through Arc + Lanesection offset (Poly3)?` I am not very clear about your question. Can you clarify?

sorry, I can't upload a file or img. I just send a sample road. You can make it a xodr file into the LGSVL. you will see. The road is combine by planview and laneoffset. Planview is basic curve line, laneoffset is a poly3 to offset the curve to polyfit the real road boundry. I don't know whether you can understand, or you can give me your email, I can tell you some detail privately.