lgsvl/simulator #1192: Fail to run multiple simulation with one python script

Yimags, 2020-12-28:
Hello all,

I am now trying to run the simulation for multiple times in one PythonAPI script. I define the number of simulation I want to run in total and add in the callback function `sim.stop()` to stop the current simulation when collision occurs. However, it get stuck and fail to start the next simulation automatically when the last one stops. Could you give me some suggestions about where the problem could take place? Thx in advance!

EricBoiseLGSVL, 2020-12-28:
@Yimags Hello, please format and place any relevant code you created here so we can debug, thanks.

Yimags, 2020-12-28:
Thx very much for the response!
I have just located the problem. `sim.stop` seems not cleaning the remains of last simulation. I add a `sim.close`. Then it works well.

closed 2020-12-28 by Yimags