lgsvl/simulator #1193: import and export HD map? cannot import traffic lights? why?

Addi2020, 2020-12-29:
I have created traffic lights in unity and export HD map (base_map.bin file), i convert base_map.bin to base_map.txt file to verify the traffic lights , it shows the traffic light are created But when I reload to Unity mean import the same file (base_map.bin) to unity for further modification, it dose not import traffic lights. it only import road lanes, boundaries and junctions.

EricBoiseLGSVL, 2020-12-29:
Did you check the box to create them on import?

Addi2020, 2020-12-30:
yes, i checked the box. it give me the following error when i import .bin file.

[image]

EricBoiseLGSVL, 2020-12-30:
Yes, this means the import failed.  Did you have any warnings or errors when you exported from lgsvl sim?

Addi2020, 2021-01-03:
no, i did not receiver any warring or error while exporting HD map. as seen below.

Successfully generated and exported Apollo HD Map!
UnityEngine.Debug:Log(Object)
Simulator.Editor.ApolloMapTool:Export(String) (at C:/Users/adnan/Desktop/sim-bigloop3d/Assets/Scripts/Editor/ApolloMapTool.cs:2834)
MapExport:OnGUI() (at C:/Users/adnan/Desktop/sim-bigloop3d/Assets/Scripts/Editor/Map/MapExport.cs:146)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr)

EricBoiseLGSVL, 2021-01-04:
This might be an edge case in the HDmap that we haven't seen.  Are you able to post the bin file here?  @luqiang21 any idea on this?  I know we have a bunch of fixes in the next release, this might be addressed already.

luqiang21, 2021-01-04:
As @EricBoiseLGSVL said, @Addi2020 Can you post the map file, so we can reproduce and check the issues?

Addi2020, 2021-01-05:
I convert .bin file to .txt, but i cannot find any traffi clight in .txt file. 
[base_map.txt](https://github.com/lgsvl/simulator/files/5768734/base_map.txt)

I dont know how to send .bin file here, when i upload .bin file to dreamview it shows all the traffic lights that i have created.

closed 2021-07-29 by Addi2020

EricBoiseLGSVL, 2021-07-29:
You can post a bin file here by zipping it.  Is this the case with the latest simulator?  We have had multiple fixes since this issue was posted.  Since you closed it, did you find a solution?

Addi2020, 2021-07-31:
Yes, Its working fine, now it can also import Traffic lights. when I import .bin file to Unity Hub 2.4.3 (not simulator) for modification, sometime It still have some missing components like traffic lights but mostly it works fine.
I use Unity Hub 2.4.3.

EricBoiseLGSVL, 2021-08-02:
Yes, some import types are needing more work and we hope to improve in the next few releases.  Don't worry about UnityHub version.  It's just a launcher and organizer of projects.  The focus should be on Unity Editor version and simulator version.