lgsvl/simulator #1194: No module named 'lgsvl' when running simulation in docker

Yimags, 2020-12-29:
Hello all,

I ran the simulation with apollo and wanna get data from cyberRT for PythonAPI. I tried to combine PythonAPI script with `listener.py` and ran it in docker. But it gives a `No module named 'lgsvl'` error. How could I fix this?

ntutangyun, 2020-12-30:
it's probably mot related to docker. it's just your python environment does not have the lgsvl PythonAPI installed. 

you could try to install by following: 

cd PythonAPI
pip install .

Yimags, 2020-12-30:
Thx for the quick response!
Actually in works well when I run python standalone. However the error takes place when I build it with bazel.
[image]

hadiTab, 2021-01-04:
Your PythonApi scripts shouldn't be placed inside the Apollo docker container as it doesn't have the Api installed by default.

Yimags, 2021-01-05:
Then I can't merge `listener.py` with Python script outside the docker which controls the simulation. How could I make use of the real time data from bridge during simulation?

closed 2021-02-01 by Yimags