lgsvl/simulator #1196: AssetBundles files are zero-length

bord81, 2021-01-02:
Hello!
I have following setup:

- Windows 10
- Unity 2019.3.15.f1

In the options I selected only Windows support.
The issue is that after doing WebUI/main build the AssetBundles files, like environment files or vehicle file have zero length and thus cannot be used by the simulator.
I haven't done anything special to assets after cloning Git repos, but I followed all steps for building the simulator as per instructions.
Do I miss some preparation steps with assets?

EricBoiseLGSVL, 2021-01-04:
You must have Linux build module installed.  You need both Windows and Linux so the bundles are cross platform.

bord81, 2021-01-05:
Yes, after enabling Linux support it worked without any issues.
Thank you very much!

closed 2021-01-05 by bord81