lgsvl/simulator #1198: Cannot run the Python API anymore

RamoramaInteractive, 2021-01-03:
I was about to run a Python script for the LGSVL simulator (Version from November 2019), but if I run a script, the windows of the simulator still shows "API ready" and nothing is happening in the command shell. It still shows empty output lines.

I'm not sure what I did wrong or what was outdated. 
I've got Python 3.7 64-bit.

Please, if you need more information let me know.

EricBoiseLGSVL, 2021-01-04:
Are you opening the terminal in the folder that holds your python scripts?  I recommend re-installing to make sure you have the latest libraries.  Did you also pull the latest from the repo?  Right @hadiTab ?

hadiTab, 2021-01-04:
@RamoramaInteractive what script were you trying to run? Can you share it or at least share some details on what it is trying to do?
Overall, PythonApi has a new release with every new simulator release so you will want to make sure you have the correct version, but I wouldn't expect anything breaking like this if you use the incorrect version (just some feature mismatch).