lgsvl/simulator #1199: How to communicate with listener.py in docker from PythonAPI outside docker?

Yimags, 2021-01-03:
Hello all,

I ran lgsvl together with apollo-6.0 and managed to get data from cyberRT with `listener.py` provided by apollo. But how could I send these data to the PythonAPI script  which I used to control the simulator outside the docker?

hadiTab, 2021-01-04:
@Yimags can you provide more details on what you are trying to achieve? Unless I'm mistaken `listener.py` is just a generic example for subscribing to a Cyber channel. That won't be relavent to controlling the vehicle in the simulator.

Yimags, 2021-01-05:
I have modified the `listener.py` a bit so that it could save data locally during simulation.

Actually I am trying to collect and save the sensor data during the simulation in a more automatic way. For example, `listener.py` starts automatically as simulation starts. When ego vehicle reaches certain position or velocity, the callback function rises then stops the simulation and the process of data collection by `listener.py` as well.

closed 2021-02-24 by Yimags