lgsvl/simulator #1200: Apollo stack unable to receive messages generated by LGSVL

vin-cloud, 2021-01-05:
I have setup my environment as per instructions detailed in LGSVl documentation below.
a. https://www.lgsvlsimulator.com/docs/build-instructions/
b. https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/
Issue:
As seen from the picture (1) below, the simulator seems to have established a connection with Apollo, but the messages generated by the simulator LGSVL is not received by the Apollo stack(Build: Passed) as seen in picture (2). Kindly help!

1.   
[image]
2.
[image]
3.
[image]
4.
[image]
5.
[image]

EricBoiseLGSVL, 2021-01-06:
@luqiang21 @hadiTab any idea or needed logs?

lemketron, 2021-01-06:
Is this all running on the same machine, with the simulator configured to connect to the bridge on localhost? It's almost as if the simulator is somehow connecting to a different bridge. It looks like the bridge is running correctly but once you've started the simulator (pressing "play" which shows the "pause" button like in your screen shot) some of the topics in cyber_monitor should immediately turn green.

vin-cloud, 2021-01-06:
> Is this all running on the same machine, with the simulator configured to connect to the bridge on localhost? It's almost as if the simulator is somehow connecting to a different bridge. It looks [...]

Yes, the entire package(simulator and Apollo stack) is running on the same machine! That's the main concern, that the topics when pressed play on simulator should have turned green, unlike the result above!

lemketron, 2021-01-06:
Can you share the Player.log file for the simulator and also the /apollo/data/log/cyber_bridge.INFO log file?

If everything has been running for a while you might want to try shutting down, rebooting, and starting everything again. The log files should be smaller too.

vin-cloud, 2021-01-06:
> Can you share the Player.log file for the simulator and also the /apollo/data/log/cyber_bridge.INFO log file?
> 
> If everything has been running for a while you might want to try shutting down, rebooting, and starting everything again. The log files should be smaller too.

There seems to be 'connection refused' statement within the Player.log file
1. player.log 
[Player.log](https://github.com/lgsvl/simulator/files/5773912/Player.log)

2.cyber_bridge.INFO (PS: kindly remove the .txt extension)
file:///home/spring/cyber_bridge.INFO.txt

vin-cloud, 2021-01-06:
> Can you share the Player.log file for the simulator and also the /apollo/data/log/cyber_bridge.INFO log file?
> 
> If everything has been running for a while you might want to try shutting down, rebooting, and starting everything again. The log files should be smaller too.

There seems to be 'connection refused' statement within the Player.log file
Well, have tried the shutdown/reboot routine with re-installation from scratch!

1. Player.log
[Player.log](https://github.com/lgsvl/simulator/files/5773921/Player.log)

2.cyber_bridge.INFO 
[cyber_bridge.log.](https://github.com/lgsvl/simulator/files/5773926/cyber_bridge.log)

Thanks!

EricBoiseLGSVL, 2021-01-06:
Yes, looks like you can't connect.  Do you have any other software using the same ports?  Do you have software managing connection permissions?

vin-cloud, 2021-01-07:
> Yes, looks like you can't connect. Do you have any other software using the same ports? Do you have software managing connection permissions?

I am not using any other software on the same port!

EricBoiseLGSVL, 2021-01-07:
Any idea @lemketron ?  This looks like an issue with apollo

lemketron, 2021-01-07:
> Any idea @lemketron ? This looks like an issue with apollo

I don't know if it's an apollo issue or docker connectivity. It's interesting that the initial connection from LGSVL to the cyber_bridge appears to succeed but then packets don't seem to flow afterwards. We're unable to reproduce this, and I'm not sure how to troubleshoot it. @martins-mozeiko might have a suggestion?

hadiTab, 2021-01-07:
@vin-cloud I don't think this is a simulator/bridge issue. I suspect that Apollo is not starting up properly. Some of those cyber channels are published directly from Apollo and do not need any bridge connection to work (for example `apollo/monitor`). I suspect that Apollo was not built successfully on your machine. I'd suggest trying a clean rebuild. You could also dig through the logs located in data/logs/ to see if you find anything as well.

vin-cloud, 2021-01-08:
> @vin-cloud I don't think this is a simulator/bridge issue. I suspect that Apollo is not starting up properly. Some of those cyber channels are published directly from Apollo and do not need any [...]

Thanks! I have actually tried deleting the docker files and then build it (recommended [Apollo 5.0](https://www.lgsvlsimulator.com/docs/apollo5-0-instructions/)) from scratch, the only option i haven't tried yet is to format on OS level and retry!

vin-cloud, 2021-07-07:
Re-installation from scratch was the only option! Closing the issue!
Thanks!

closed 2021-07-07 by vin-cloud